YosysHQ/yosys #2321: 'std::out_of_range' at 'dict::at()` during techmap with shift logic loop

The6P4C, 2020-08-02:
## Steps to reproduce the issue
Run `yosys -p "opt_clean; techmap" top.v` with what is an obviously badly written piece of Verilog:

```verilog
// top.v
module top(output x);
        assign x = 1 >> x;
endmodule
```

## Expected behavior
Yosys reports an error due to the logic loop, rather than crashing.

## Actual behavior
An `std::out_of_range` is raised during the techmap pass.

```
3.2. Continuing TECHMAP pass.
terminate called after throwing an instance of 'std::out_of_range'
  what():  dict::at()
Aborted (core dumped)
```

(thank you to @Ravenslofty for helping minimise)

mwkmwkmwk, 2020-08-02:
Reduced further:

```
$ cat m.v
module m (input i, output o);
wire [1023:0] _TECHMAP_DO_00_ = "CONSTMAP; ";
endmodule
$ cat top.v
module top(output o);
m m (.o(o), .i(o));
endmodule
$ yosys -p 'techmap -map m.v' top.v
```

mwkmwkmwk, 2020-08-05:
Hilariously, this fix actually optimizes away the whole shift into nothingness — it is at some point optimized to just a `$_NOT_` cell as would be expected if the input and output were actually different wires, but then the "double inversion" rule triggers in `opt_expr` — since `x` is now an output of an inverter, whose input (also `x`) is also an output of an inverter (the same damn inverter), the first inverter is removed, and its output (`x`) is connected to the second inverter's input (also `x`)...

PR #2327 (2020-08-20): techmap.CONSTMAP: Handle outputs before inputs.
Fixes #2321.
commit b4a4cb08 (2020-08-20): techmap.CONSTMAP: Handle outputs before inputs.

Fixes #2321.
closed 2020-08-20 by clairexen