YosysHQ/yosys #2343: xaiger failure during abc9 pass

gsomlo, 2020-08-16:
## Steps to reproduce the issue

See [gls_mvce.tar.xz](https://github.com/gsomlo/yosys-abc-test/blob/master/gls_mvce.tar.xz). Reproduce with:
```
tar xfJ gls_mvce.tar.xz
cd gls_mvce
bash RunMe.sh
```

## Expected behavior

yosys successfully completes synthesis run, including abc9 techmapping and aiger stage

## Actual behavior

yosys fails with:
```
8.47.18.6. Executing AIGER frontend.
/usr/include/c++/10/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::State>::entry_t; _Alloc = std::allocator<Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::State>::entry_t>; std::vector<_Tp, _Alloc>::reference = Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::State>::entry_t&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
```

Reproducer tarball includes a backtrace in `yosys-abc-QNqgGU-backtrace.txt` and the entire yosys-abc temp folder `yosys-abc-QNqgGU`.

I am using upstream abc at commit [448f263](https://github.com/berkeley-abc/abc/commit/448f26344325cc78942d576740457036671b0976) ***with [PR 89](https://github.com/berkeley-abc/abc/pull/89) applied on top*** (which gets rid of a problem whereby abc itself crashes during the abc9 techmapping stage, see [issue 84](https://github.com/berkeley-abc/abc/issues/84). Given the recent commit version of abc I'm using, I believe these problems are also shared with the [current "vendored" abc fork maintained by yosyshq](https://github.com/yosyshq/abc).

For context, the verilog files in the reproducer are building a LiteX + RocketChip SoC.

Happy to provide any extra details to help further narrow down the problem. Thanks!

Ravenslofty, 2020-08-17:
cc @eddiehung

eddiehung, 2021-08-31:
No longer reproduces on vanilla b2e9717419e9a852f4e64f12891b8e9742900917 -- the attached testcase gets all the way through the flow. Closing for now.

closed 2021-08-31 by eddiehung