YosysHQ/yosys #2373: CXXRTL: Don't force combinatorial outputs of modules to be a wire object

tomverbeure, 2020-09-04:
Right now, all outputs of a modules are forced to be a wire object, even if that output is a combinatorial signal instead of coming out of a FF.

For example, the assignment to `led` in [this code](https://github.com/tomverbeure/cxxrtl_eval/blob/6516b811e66b56022ef14bd6fbd00fedf8b18130/blink_basic/blink.v#L9), results in the following generated code:

```cpp
struct p_blink : public module {
	// \init: 0
	// \src: blink.v:4.16-4.23
	wire<12> p_counter {0u};
	// \src: blink.v:2.32-2.35
	/*output*/ wire<1> p_led;                            // <<<<<<<<<<<<<<<<< wire, not value!
	// \src: blink.v:2.20-2.23
	/*input*/ value<1> p_clk;
	value<1> prev_p_clk;
	bool posedge_p_clk() const {
		return !prev_p_clk.slice<0>().val() && p_clk.slice<0>().val();
	}

	bool eval() override;
	bool commit() override;
	void debug_info(debug_items &items, std::string path = "") override;
}; // struct p_blink
``` 

It also leads to the following Yosys warning, and wasted simulation cycles:
```
Module `blink' contains buffered combinatorial wires:
  led
Warning: Design contains buffered combinatorial wires, which require delta cycles during evaluation.
```

Since CXXRTL has get and set methods, it's not necessary anymore to force outputs to be wires for consistency, so it's better to make them values.

whitequark, 2020-11-02:
This was actually fixed in https://github.com/YosysHQ/yosys/pull/2149. However, the changes in that PR do not unbuffer wires in your example because it has no top module. Since it is not obvious that a top module is necessary here, and to make the defaults more friendly, I added an invocation of `hierarchy -auto-top` to the preparation done in the CXXRTL backend in https://github.com/YosysHQ/yosys/pull/2426, after which your example works as expected.

tomverbeure, 2020-11-02:
> However, the changes in that PR do not unbuffer wires in your example because it has no top module.

And with that, a major mystery is solved: I just couldn't understand why some of my examples had a `wire` while others had a `value`, even if both of them had a combinatorial output!

whitequark, 2020-11-02:
Yup. Unbuffered outputs strictly speaking do not *have* to be restricted to top modules, but it's convenient because it lets us avoid complicating internal module-module interfaces in hierarchical designs.

PR #2426 (2020-11-02): cxxrtl: run `hierarchy -auto-top` if no top module is present
In most cases, a CXXRTL simulation would use a top module, either because this module serves as an entry point to the CXXRTL C API, or because the outputs of a top module are unbuffered, improving performance. Taking this into account, the CXXRTL backend now runs `hierarchy -auto-top` if there is no top module. For the few cases where this behavior is unwanted, it now accepts a `-nohierarchy` option.

Fixes #2373.
commit 65083e95 (2020-11-02): cxxrtl: run `hierarchy -auto-top` if no top module is present.

In most cases, a CXXRTL simulation would use a top module, either
because this module serves as an entry point to the CXXRTL C API,
or because the outputs of a top module are unbuffered, improving
performance. Taking this into account, the CXXRTL backend now runs
`hierarchy -auto-top` if there is no top module. For the few cases
where this behavior is unwanted, it now accepts a `-nohierarchy`
option.

Fixes #2373.
closed 2020-11-02 by whitequark