YosysHQ/yosys #2374: CXXRTL: assert on combinational feedback code

tomverbeure, 2020-09-04:
## Steps to reproduce the issue

[endless_toggle.zip](https://github.com/YosysHQ/yosys/files/5177255/endless_toggle.zip)

Verilog design with combinational feedback loop:

```verilog
module endless_toggle(input clk1);

    wire toggle = 0;
    wire taggle;

    assign taggle = ~toggle;
    assign toggle = taggle;

endmodule
```

```
yosys -p "read_verilog endless_toggle.v; hierarchy -top endless_toggle; write_cxxrtl endless_toggle.cpp"
```

## Expected behavior

CXXRTL does something that doesn't crash. Error message or C code.

## Actual behavior

```
ERROR: Assert `flow.wire_comb_defs[wire].size() == 1' failed in backends/cxxrtl/cxxrtl_backend.cc:2185.
```

It's obviously a nonsense design, but chances are that this assert would trigger valid designs as well.

tomverbeure, 2020-09-04:
When removing the `= 0` in the `wire toggle = 0;`, it compiles fine.

whitequark, 2020-09-04:
Oh right, that's invalid Verilog, it has multiple drivers for `toggle`.

whitequark, 2020-09-04:
I guess I can still add a check for that, not too hard.

tdaede, 2020-09-13:
I just got the same assert with cxxrtl and ao68000: https://gitlab.com/Chips4Makers/ao68000

It's possible this design has such a feedback loop, I'll see if I can find any obvious ones.

tdaede, 2020-09-22:
I found a minimal reproducer for my assert. The problem is code I added to initialize some memories which reuses the same integer for both initialization loops. Removing the second loop, or adding a second integer j and using that for the second loop, avoids the assert.

```
module memory_registers(

);

// Generic version of A register bank
reg [31:0] An[0:7];

integer i;
initial begin
    for (i = 0; i < 8; i = i + 1) begin
        An[i] = 0;
    end
end

// Generic version of D register bank
reg [31:0] Dn[0:7];

initial begin
    for (i = 0; i < 8; i = i + 1) begin
        Dn[i] = 0;
    end
end

endmodule
```

whitequark, 2020-11-01:
> Removing the second loop, or adding a second integer j and using that for the second loop, avoids the assert.

Yeah, this module is driving `i` twice. Your issue is arguably a frontend bug though.

whitequark, 2020-11-01:
Will be fixed by https://github.com/YosysHQ/yosys/pull/2424 and print:

```
ERROR: Wire endless_toggle.toggle has multiple drivers.
```

PR #2424 (2020-11-01): cxxrtl: don't assert on wires with multiple drivers
Fixes #2374.
commit cdf4ce98 (2020-11-01): cxxrtl: don't assert on wires with multiple drivers.

Fixes #2374.
closed 2020-11-01 by whitequark