YosysHQ/yosys #2375: CXXRTL RFE: option to specify maximum number of delta cycles to avoid runaway sims.

tomverbeure, 2020-09-04:
It'd be useful to abort the simulation when you reach a certain maximum number of delta cycles in eval loops.

This could either be specific on the `write_cxxrtl` command line, or something that's specified in the API. I'd vote for it to be enabled by default, but I don't have a strong opinion on that.

When reaching the maximum number, it should print out the hierarchical path where it did (though with default settings, that wouldn't really help.)

whitequark, 2020-09-04:
> It'd be useful to abort the simulation when you reach a certain maximum number of delta cycles in eval loops.

I'm not convinced that there is a lot of value in this. The only case where you'd need this functionality is non-synthesizable code with combinatorial feedback loops, which you really shouldn't write in first place.

whitequark, 2020-09-04:
Oh, and if you find yourself needing this functionality often, you can always just implement `step` yourself--it's not special in any way.

tomverbeure, 2020-09-05:
> I'm not convinced that there is a lot of value in this. The only case where you'd need this functionality is non-synthesizable code with combinatorial feedback loops, which you really shouldn't write in first place. This would make the default `step` function harder to understand and slower.

As for non-synthesizable code: I use CXXRTL as my main simulator now, which means that I use it a lot before I ever synthesize. Most linting tools don't catch combinational loops, verilator doesn't catch it either.

I don't think the cost of a single increment and compare per eval cycle will make the simulation slower. Compared to everything else in `cxxrtl.h`, `step()` is probably the simplest function. I think people would manage.

closed 2020-09-05 by tomverbeure

whitequark, 2020-09-05:
I guess. For the most part, I don't like having an arbitrary limit that needs to be made configurable, threaded through arguments, exposed in the C API, and so on. I would dislike it less as a macro, but it would also be harder to use that way.

tomverbeure, 2020-09-05:
What about somethingn like `size_t step(int max_delta_cycles = -1) {` ?

It retains the current behavior, removes the need for a command line option of `write_cxxrtl`, and it's a very localized change to only `step()` and nothing else?

For the C API, a `size_t cxxrtl_step_with_limits(cxxrtl_handle handle, int max_delta_cycles) {` is admittedly not very elegant.