YosysHQ/yosys #2377: Feature request: support SV array query functions like $high and $left

dsvf, 2020-09-11:
Systemverilog has functions that return information about the bounds of a vector, like `$high/$low` and `$left/$right`. They are convenient helper functions for width-agnostic code/functions.

To give an example, in my use case they are extremely helpful for fixpoint DSP, e.g. automatically calculating the correct vector width and tracking the number of integer/fractional bits:
```verilog
logic [3:-4] a; // a Q4.4 number
logic [3:-8] b; // a Q4.8 number
logic [$high(a)+$high(b)-1 : $low(a)+$low(b)] c = a*b; // automatically generates a 20-bit wide Q8.12 number

logic [$high(c) : 0] c_integer = c[$high(c) : 0];    // extracting the integer part is easy since both bounds can be obtained from a given vector
```

I see no way whatsoever to achieve this with the SV subset supported by Yosys; `read_verilog -sv` fails with `ERROR: Can't resolve function name ``\$left'`

udif, 2020-09-13:
A few years ago I did the $bits() and $size() PR. I think $high() and $low() are pretty easy to do and are quite similar.
(reference: https://github.com/YosysHQ/yosys/pull/425)
I wasn't even aware that they were not implemented yet.
I was quite busy recently, but if no one else pick this up, I'll try to do it in a few days.

udif, 2020-09-15:
Have a go: https://github.com/YosysHQ/yosys/pull/2378

udif, 2020-10-01:
PR accepted!
https://github.com/YosysHQ/yosys/commit/7e2fc2eaeb70179c8da3e5dc8be800f486d5b912

mwkmwkmwk, 2020-10-02:
@udif for future reference: please write "Fixes #2377" in the commit message in cases like this — it will automatically link the PR with the issue, and merging the PR will then automatically also close the issue.

I suppose this issue can now be closed?

closed 2020-12-05 by whitequark