YosysHQ/yosys #2385: make test results in Test: arrays02 -> ERROR!

ckqee, 2020-09-23:
## Steps to reproduce the issue

Doing the following three steps does not work having already got iverilog as well:
```
  git clone https://github.com/YosysHQ/yosys
  cd yosys
  make test
```

```
[user7@ yosys]$ which iverilog
~/software/iverilog/bin/iverilog
[user7@ yosys]$ which iverilog

[user7@ yosys]$ iverilog -V
Icarus Verilog version 11.0 (devel) (s20150603-853-g4466a19)

Copyright 1998-2020 Stephen Williams

[user7@ yosys]$ git status
# On branch master
nothing to commit, working directory clean
[user7@ yosys]$ date
Wed Sep 23 07:23:48 PDT 2020
[user7@ yosys]$ git pull
Already up-to-date.
[user7@ yosys]$ date
Wed Sep 23 07:24:09 PDT 2020
[user7@ yosys]$ make test
.
.
.
[100%] Building share/sf2/cells_map.v
[100%] Building share/sf2/cells_sim.v
cd tests/simple && bash run-test.sh ""
make[1]: Entering directory `/home/user7/software/yosys/tests/simple'
+ gcc -Wall -o /home/user7/software/yosys/tests/tools/cmp_tbdata /home/user7/software/yosys/tests/tools/cmp_tbdata.c
Test: arrays02 -> ERROR!
Note: Make sure that 'iverilog' is an up-to-date git checkout of Icarus Verilog.
make[1]: *** [arrays02.sv] Error 1
make[1]: Leaving directory `/home/user7/software/yosys/tests/simple'
make: *** [test] Error 2
```

*Provide instructions for reproducing the issue. Make sure to include
all necessary source files. (You can simply drag&drop a .zip file into
the issue editor.)*

Also, make sure that the issue is actually reproducible in current git
master of Yosys.

See https://stackoverflow.com/help/mcve for some information on how to
create a Minimal, Complete, and Verifiable example (MCVE).

Please do not waste our time with issues that lack sufficient information
to reproduce the issue easily. We will simply close those issues.

Contact https://www.symbioticeda.com/ if you need commercial support for Yosys.

## Expected behavior

```
Not sure what it should be
```

*Please describe the behavior you would have expected from the tool.*

## Actual behavior

```
[100%] Building share/sf2/cells_map.v
[100%] Building share/sf2/cells_sim.v
cd tests/simple && bash run-test.sh ""
make[1]: Entering directory `/home/user7/software/yosys/tests/simple'
+ gcc -Wall -o /home/user7/software/yosys/tests/tools/cmp_tbdata /home/user7/software/yosys/tests/tools/cmp_tbdata.c
Test: arrays02 -> ERROR!
Note: Make sure that 'iverilog' is an up-to-date git checkout of Icarus Verilog.
make[1]: *** [arrays02.sv] Error 1
make[1]: Leaving directory `/home/user7/software/yosys/tests/simple'
make: *** [test] Error 2
```

*Please describe how the behavior you see differs from the expected behavior.*

nakengelhardt, 2020-09-24:
I have encountered this error before and it only happens occasionally immediately after building yosys. If I just run `make test` again, it passes the second time. Unfortunately I've never been able to reproduce it reliably enough to figure out what causes it.

Do you get the error every time? Do you also get it if you manually run `cd tests/simple && bash run-test.sh`?

nakengelhardt, 2020-09-24:
Also, to help debug this, can you show the contents of `yosys/tests/simple/arrays02.log` and `yosys/tests/simple/arrays02.out` (as a zip)?

trayres, 2023-04-07:
I just ran into this issue; running 
`make test`
a second time didn't fix it. 
[arrays02.out.zip](https://github.com/YosysHQ/yosys/files/11180100/arrays02.out.zip)

It appears the only file in arrays02.out for me is "arrays02_ref.sv"; I'm incl
[arrays02.out.zip](https://github.com/YosysHQ/yosys/files/11180107/arrays02.out.zip)
uding it inline as well because it is short:
```verilog
module uut_arrays02(clock, we, addr, wr_data, rd_data);

input clock, we;
input [3:0] addr, wr_data;
output [3:0] rd_data;
reg [3:0] rd_data;

reg [3:0] memory [16];

always @(posedge clock) begin
	if (we)
		memory[addr] <= wr_data;
	rd_data <= memory[addr];
end

endmodule
```

mfatimah, 2024-04-30:
I have had similar problems, here is my logfile

[arrays02.log](https://github.com/YosysHQ/yosys/files/15168052/arrays02.log)