YosysHQ/yosys #2388: shiftx semantics

renau, 2020-09-26:
I am a bit confused with the shiftx semantics. I have this two modules and they fail LEC.

(This is more a topic for a dicussion/question channel like gitter, but I do not see any for yosys).

Gold model:
```
module gold_model(input [7:0] a, input [2:0] b, input [1:0] b2, output [3:0] c);
  assign c = a[$signed(b) +: 4];
endmodule
```

Shift based implemtation:
```
module gate_mode(input [7:0] a, input [2:0] b, input [1:0] b2, output [3:0] c);

  reg [3:0] tmp;
  assign c = tmp;
  reg [3:0]  bneg;

  always @(a or b or b2) begin
    bneg = -$signed(b);
    if ($signed(b)<0) begin
      tmp = (a<<bneg) | ((4'bx)&((1<<(-$signed(b)))-1)); // the | 4'bx.... part is not needed just for VCS match
      //UNCOMMENT to pass formal: tmp = a[$signed(b) +: 4];
    end else begin
      tmp = a>>($signed(b));
    end
  end

endmodule
```

After running lec with yosys, I get:
```   ______                   ___       ___       _ _            _ _
  (_____ \                 / __)     / __)     (_) |          | | |
   _____) )___ ___   ___ _| |__    _| |__ _____ _| | _____  __| | |
  |  ____/ ___) _ \ / _ (_   __)  (_   __|____ | | || ___ |/ _  |_|
  | |   | |  | |_| | |_| || |       | |  / ___ | | || ____( (_| |_
  |_|   |_|   \___/ \___/ |_|       |_|  \_____|_|\_)_____)\____|_|

  Time Signal Name             Dec       Hex           Bin
  ---- --------------- ----------- --------- -------------
     1 \gate_c                  --        --          0xxx
     1 \gold_c                  --        --          x000
     1 \in_a                     0         0      00000000
     1 \in_b                     5         5           101
     1 \in_b2                    0         0            00
     1 \trigger                  1         1             1
```

The yosys command is:
```
              design -copy-from gold -as gold gold
              design -copy-from gate -as gate gate

              miter -equiv -flatten -make_outputs -ignore_gold_x gold gate miter
              hierarchy -top miter
              write_verilog miter.v
              sat -tempinduct -prove trigger 0 -set-init-undef -enable_undef -set-def-inputs -ignore_unknown_cells -show-ports miter
```

I also have access to VCS, when I have this top level driver. I see that both generate the same results:
```
module top();

  reg [7:0] a;
  reg [2:0] b;
  reg [3:0] b2;
  wire [3:0] c;
  wire [3:0] cx;

  initial begin

    //a = 8'bx010x000;
    a = 8'b000x010x;

    b2 = 4;

    for(b2=0;b2<8;b2=b2+1) begin
      b = b2;
      #1 $display("a:%b b:%d c:%b cx:%b",a,$signed(b),c,cx);
    end
  end

  gate_model  i(a,b,c2,c);
  gold_model j(a,b,c2,cx);

endmodule
````

It generates this VCS result: (exact match)
```
a:000x010x b:          0 c:010x cx:010x
a:000x010x b:          1 c:x010 cx:x010
a:000x010x b:          2 c:0x01 cx:0x01
a:000x010x b:          3 c:00x0 cx:00x0
a:000x010x b:         -4 c:xxxx cx:xxxx
a:000x010x b:         -3 c:xxxx cx:xxxx
a:000x010x b:         -2 c:0xxx cx:0xxx
a:000x010x b:         -1 c:10xx cx:10xx
```

The 4'bx part is not needed. THis should be enough, but it was to make it match even with x's
```
    if ($signed(b)<0) begin
      tmp = (a<<bneg); /// The rest is just for VCS | ((4'bx)&((1<<(-$signed(b)))-1));
    end else begin
      tmp = a>>($signed(b));
    end
```

Is the SAT solver for the shiftx gate wrong? What is the semantic different? (Trying all options in vcs, they always match, so not sure why the SAT solver says to fail. Trying the failing inputs with vcs does not missmatch).

renau, 2020-09-26:
The latest (git) version of yosys does not have this issue. I guess that problem solved.

closed 2020-09-26 by renau

Xiretza, 2020-09-26:
> (This is more a topic for a dicussion/question channel like gitter, but I do not see any for yosys).

For future reference: the [#yosys IRC channel](http://webchat.freenode.net/?channels=yosys) on freenode is the official chat platform, as linked to on http://www.clifford.at/yosys/.

> The latest (git) version of yosys does not have this issue. I guess that problem solved.

Actually, I can reproduce the problem on latest master (cd8b2ed4) with the two verilog files you provided - the following script does indeed fail:

```yosys
miter -equiv -flatten -make_outputs gold_model gate_mode miter
hierarchy -top miter
write_verilog miter.v
sat -verify -prove trigger 0 -set-init-undef -set-def-inputs -show-ports miter
```

```
$ ./yosys shiftx_gate.v shiftx_gold.v shiftx_test.ys
[...]
 Yosys 0.9+3586 (git sha1 cd8b2ed4, gcc 10.2.0 -fPIC -Os)
[...]
SAT proof finished - model found: FAIL!

   ______                   ___       ___       _ _            _ _ 
  (_____ \                 / __)     / __)     (_) |          | | |
   _____) )___ ___   ___ _| |__    _| |__ _____ _| | _____  __| | |
  |  ____/ ___) _ \ / _ (_   __)  (_   __|____ | | || ___ |/ _  |_|
  | |   | |  | |_| | |_| || |       | |  / ___ | | || ____( (_| |_ 
  |_|   |_|   \___/ \___/ |_|       |_|  \_____|_|\_)_____)\____|_|

  Signal Name             Dec       Hex           Bin
  --------------- ----------- --------- -------------
  \gate_c                  --        --          xxxx
  \gold_c                   0         0          0000
  \in_a                     0         0      00000000
  \in_b                     0         0           000
  \in_b2                    0         0            00
  \trigger                  1         1             1

ERROR: Called with -verify and proof did fail!
```

Trying these inputs with `eval` gives a hint:

```
$ ./yosys shiftx_gold.v -p 'eval -set-undef -set a 0 -set b 0'
[...]
Eval result: \c = 4'0000.
```

```
$ ./yosys shiftx_gate.v -p 'eval -set-undef -set a 0 -set b 0'
[...]
Failed to evaluate signal \c: Missing value for \tmp. -> setting to undef
Eval result: \c = 4'xxxx.
```

It seems that neither `eval` nor `sat` can handle processes, with `sat` ignoring them completely silently - adding a `proc` pass before either solves all these problems:

```yosys
proc gate_mode
miter -equiv -flatten -make_outputs gold_model gate_mode miter
hierarchy -top miter
write_verilog miter.v
sat -verify -prove trigger 0 -set-init-undef -set-def-inputs -show-ports miter
```

```
$ ./yosys shiftx_gate.v shiftx_gold.v -s shiftx_test.ys
[...]
SAT proof finished - no model found: SUCCESS!

                  /$$$$$$      /$$$$$$$$     /$$$$$$$    
                 /$$__  $$    | $$_____/    | $$__  $$   
                | $$  \ $$    | $$          | $$  \ $$   
                | $$  | $$    | $$$$$       | $$  | $$   
                | $$  | $$    | $$__/       | $$  | $$   
                | $$/$$ $$    | $$          | $$  | $$   
                |  $$$$$$/ /$$| $$$$$$$$ /$$| $$$$$$$//$$
                 \____ $$$|__/|________/|__/|_______/|__/
                       \__/                              
```