YosysHQ/yosys #2389: Yosys stucks in an infinite loop when running synthesis

nalzok, 2020-09-26:
## Steps to reproduce the issue

1. Download [oled.zip](https://github.com/YosysHQ/yosys/files/5286479/oled.zip)
2. Unzip it and `cd` into the directory
3. `make`

## Expected behavior

Some background first: I am trying to use Digilent's [Pmod OLED](https://reference.digilentinc.com/reference/pmod/pmodoled/start) with [icesugar](https://github.com/wuxx/icesugar), an iCE40UP5k-powered development board by adapting the [Nexys 3 Verilog Example - ISE 14.2](https://reference.digilentinc.com/_media/reference/pmod/pmodoled/pmodoledctrl_verilog.zip) example. Chances are that I made some mistake so the attached code won't work, but I still expect yosys to either

1. produce a JSON netlist on success, or
2. emit an error on failure.

## Actual behavior

It appears to stuck in an infinite loop. I have waited for more than half an hour and it didn't output anything. The last few log lines are

```
6.3. Executing PROC pass (convert processes to netlists).

6.3.1. Executing PROC_CLEAN pass (remove empty switches from decision trees).
Cleaned up 0 empty switches.

6.3.2. Executing PROC_RMDEAD pass (remove dead branches from decision trees).
Marked 1 switch rules as full_case in process $proc$/usr/local/bin/../share/yosys/ice40/cells_sim.v:1330$335 in module SB_DFFNES.
Marked 1 switch rules as full_case in process $proc$/usr/local/bin/../share/yosys/ice40/cells_sim.v:1274$328 in module SB_DFFNESS.
Marked 1 switch rules as full_case in process $proc$/usr/local/bin/../share/yosys/ice40/cells_sim.v:1199$324 in module SB_DFFNER.
Marked 1 switch rules as full_case in process $proc$/usr/local/bin/../share/yosys/ice40/cells_sim.v:1143$317 in module SB_DFFNESR.
Marked 1 switch rules as full_case in process $proc$/usr/local/bin/../share/yosys/ice40/cells_sim.v:1074$314 in module SB_DFFNS.
Marked 1 switch rules as full_case in process $proc$/usr/local/bin/../share/yosys/ice40/cells_sim.v:1026$311 in module SB_DFFNSS.
Marked 1 switch rules as full_case in process $proc$/usr/local/bin/../share/yosys/ice40/cells_sim.v:957$308 in module SB_DFFNR.
Marked 1 switch rules as full_case in process $proc$/usr/local/bin/../share/yosys/ice40/cells_sim.v:909$305 in module SB_DFFNSR.
Marked 1 switch rules as full_case in process $proc$/usr/local/bin/../share/yosys/ice40/cells_sim.v:753$297 in module SB_DFFES.
Marked 1 switch rules as full_case in process $proc$/usr/local/bin/../share/yosys/ice40/cells_sim.v:697$290 in module SB_DFFESS.
Marked 1 switch rules as full_case in process $proc$/usr/local/bin/../share/yosys/ice40/cells_sim.v:622$286 in module SB_DFFER.
Marked 1 switch rules as full_case in process $proc$/usr/local/bin/../share/yosys/ice40/cells_sim.v:566$279 in module SB_DFFESR.
Marked 1 switch rules as full_case in process $proc$/usr/local/bin/../share/yosys/ice40/cells_sim.v:497$276 in module SB_DFFS.
Marked 1 switch rules as full_case in process $proc$/usr/local/bin/../share/yosys/ice40/cells_sim.v:449$273 in module SB_DFFSS.
Marked 1 switch rules as full_case in process $proc$/usr/local/bin/../share/yosys/ice40/cells_sim.v:380$270 in module SB_DFFR.
Marked 1 switch rules as full_case in process $proc$/usr/local/bin/../share/yosys/ice40/cells_sim.v:332$267 in module SB_DFFSR.
```

mwkmwkmwk, 2020-09-26:
It seems that the `BitPatternPool` used in `proc_rmdead` has impressively bad performance when long random bit patterns are involved.  We'll try to think of a proper solution; for a workaround, try changing the strings used in OledInit.v as state values to enumerated constants with sequential values.

nalzok, 2020-09-26:
> for a workaround, try changing the strings used in OledInit.v as state values to enumerated constants with sequential values.

It works! I had to replace the "state strings" in both `Oledinit.v` and `OledEX.v` with sequential integers to get yosys to run in a reasonable amount of time, though.

rob-ng15, 2020-11-05:
Is this the reason also for #2415 ?

PR #2895 (2021-07-30): proc_rmdead: use explicit pattern set when there are no wildcards
If width of a case expression was large, explicit patterns could cause the existing logic to take an extremely long time, or exhaust the maximum size of the underlying set. For cases where all of the patterns are fully defined and there are no constants in the case expression, this change uses a simple set to track which patterns have been seen.

Feedback appreciated!

Fixes #2389
closed 2021-07-30 by zachjs