YosysHQ/yosys #2402: Command "fsm_export" does not generate a .kiss2 file to represent correspondings FSM

fanhy2114, 2020-10-12:
module counter (clk, rst, en, count);
        input clk, rst, en;
        ouput reg[1:0] count;
        
         always @(posedge clk)
                    if (rst)
                        count <= 2'd0
                   else if (en)
                        count <= count + 2'd1
endmodule

Here is my yosys script:
      read_verilog counter.v
      proc; opt; fsm -no map;
      fsm_export -o counter.kiss2

The log message said that "exporting FSMs in KISS2 file format", but i cannot find "counter.kiss2", I don't understand why,
I would appreciate it if you can tell me the reason and how to get fsm file in KISS2 format, thx :)

Here is the log message:

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Wolf <claire@symbioticeda.com>          |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+3556 (git sha1 3f27a4ea, clang  -fPIC -Os)

-- Executing script file `test.ys' --

1. Executing Verilog-2005 frontend: counter.v
Parsing Verilog input from `counter.v' to AST representation.
Generating RTLIL representation for module `\counter'.
Successfully finished Verilog frontend.

2. Executing PROC pass (convert processes to netlists).

2.1. Executing PROC_CLEAN pass (remove empty switches from decision trees).
Cleaned up 0 empty switches.

2.2. Executing PROC_RMDEAD pass (remove dead branches from decision trees).
Marked 1 switch rules as full_case in process $proc$counter.v:6$1 in module counter.
Removed a total of 0 dead cases.

2.3. Executing PROC_PRUNE pass (remove redundant assignments in processes).
Removed 0 redundant assignments.
Promoted 0 assignments to connections.

2.4. Executing PROC_INIT pass (extract init attributes).

2.5. Executing PROC_ARST pass (detect async resets in processes).

2.6. Executing PROC_MUX pass (convert decision trees to multiplexers).
Creating decoders for process `\counter.$proc$counter.v:6$1'.
     1/1: $0\count[1:0]

2.7. Executing PROC_DLATCH pass (convert process syncs to latches).

2.8. Executing PROC_DFF pass (convert process syncs to FFs).
Creating register for signal `\counter.\count' using process `\counter.$proc$counter.v:6$1'.
  created $dff cell `$procdff$8' with positive edge clock.

2.9. Executing PROC_CLEAN pass (remove empty switches from decision trees).
Found and cleaned up 2 empty switches in `\counter.$proc$counter.v:6$1'.
Removing empty process `counter.$proc$counter.v:6$1'.
Cleaned up 2 empty switches.

3. Executing OPT pass (performing simple optimizations).

3.1. Executing OPT_EXPR pass (perform const folding).
Optimizing module counter.

3.2. Executing OPT_MERGE pass (detect identical cells).
Finding identical cells in module `\counter'.
Removed a total of 0 cells.

3.3. Executing OPT_MUXTREE pass (detect dead branches in mux trees).
Running muxtree optimizer on module \counter..
  Creating internal representation of mux trees.
  Evaluating internal representation of mux trees.
  Analyzing evaluation results.
Removed 0 multiplexer ports.
<suppressed ~1 debug messages>

3.4. Executing OPT_REDUCE pass (consolidate $*mux and $reduce_* inputs).
  Optimizing cells in module \counter.
Performed a total of 0 changes.

3.5. Executing OPT_MERGE pass (detect identical cells).
Finding identical cells in module `\counter'.
Removed a total of 0 cells.

3.6. Executing OPT_DFF pass (perform DFF optimizations).
Adding SRST signal on $procdff$8 ($dff) from module counter (D = $procmux$3_Y, Q = \count, rval = 2'00).
Adding EN signal on $auto$opt_dff.cc:702:run$9 ($sdff) from module counter (D = $add$counter.v:10$2_Y, Q = \count).

3.7. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \counter..
Removed 2 unused cells and 5 unused wires.
<suppressed ~3 debug messages>

3.8. Executing OPT_EXPR pass (perform const folding).
Optimizing module counter.

3.9. Rerunning OPT passes. (Maybe there is more to do..)

3.10. Executing OPT_MUXTREE pass (detect dead branches in mux trees).
Running muxtree optimizer on module \counter..
  Creating internal representation of mux trees.
  No muxes found in this module.
Removed 0 multiplexer ports.

3.11. Executing OPT_REDUCE pass (consolidate $*mux and $reduce_* inputs).
  Optimizing cells in module \counter.
Performed a total of 0 changes.

3.12. Executing OPT_MERGE pass (detect identical cells).
Finding identical cells in module `\counter'.
Removed a total of 0 cells.

3.13. Executing OPT_DFF pass (perform DFF optimizations).

3.14. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \counter..

3.15. Executing OPT_EXPR pass (perform const folding).
Optimizing module counter.

3.16. Finished OPT passes. (There is nothing left to do.)

4. Executing FSM pass (extract and optimize FSM).

4.1. Executing FSM_DETECT pass (finding FSMs in design).

4.2. Executing FSM_EXTRACT pass (extracting FSM from design).

4.3. Executing FSM_OPT pass (simple optimizations of FSMs).

4.4. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \counter..

4.5. Executing FSM_OPT pass (simple optimizations of FSMs).

4.6. Executing FSM_RECODE pass (re-assigning FSM state encoding).

4.7. Executing FSM_INFO pass (dumping all available information on FSM cells).

5. Executing FSM_EXPORT pass (exporting FSMs in KISS2 file format).

End of script. Logfile hash: f37e6c0425, CPU: user 0.00s system 0.00s, MEM: 10.01 MB peak
Yosys 0.9+3556 (git sha1 3f27a4ea, clang  -fPIC -Os)
Time spent: 28% 3x opt_expr (0 sec), 23% 3x opt_clean (0 sec), ...

fanhy2114, 2020-10-12:
And I try this https://stackoverflow.com/questions/32645965/fsm-export-using-yosys, still, there are no .kiss2 file generated.

daveshah1, 2020-10-12:
My understanding is that Yosys has heuristics to extract certain patterns of FSMs from the RTL that might benefit from optimisation (like recoding to one-hot). A counter is best kept as an adder and a register rather than being subject to FSM optimisations, so it is never extracted and therefore never written as a .kiss2. Yosys' FSM functionality is not intended to find all possible FSMs out of a given design.

fanhy2114, 2020-10-13:
> My understanding is that Yosys has heuristics to extract certain patterns of FSMs from the RTL that might benefit from optimisation (like recoding to one-hot). A counter is best kept as an adder and [...]

Thanks, I used to believe that the Verilog code must be converted into FSM. The example in the link(https://stackoverflow.com/questions/32645965/fsm-export-using-yosys) generate FSM successfully, but they're still no FSM according to my test.  However, it is 5 years ago, the implementation may be changed.
I wonder if we can translate Verilog to time automata and verify its safety. Anyway, thank you.

Vulcan-Dynamyte, 2020-10-30:
Hey, @fanhy2114 if you still need the FSM extraction thingy. I'd recommend synthesizing your design and using SIS(v1.4  link https://jackhack96.github.io/logic-synthesis/sis.html). This is the fastest method. Be careful as this is pretty slow for huge FSM and the RAM requirement is exponential. Some universities have tools that can extract FSMs from netlists, but all of them are private. Unless you are a part of the research group you will not get access to the tools.

whitequark, 2020-10-31:
> Some universities have tools that can extract FSMs from netlists, but all of them are private. Unless you are a part of the research group you will not get access to the tools.

Can someone remind me why we, the public, pay academics?

closed 2020-12-05 by whitequark