YosysHQ/yosys #2408: Provide list of built-in plugins

umarcor, 2020-10-22:
When Yosys is built without dynamic plugin loading support, using option `-m` produces the following error:

> ERROR: This version of yosys is built without plugin support.

However, some plugins/frontends might be built-in. For example, ghdl-yosys-plugin can be built either with Yosys or as a shared library. Using `-m ghdl` in the former case produces the mentioned error. That's partially misleading because `ghdl` commands are available. Instead, I would like to propose listing the built-in plugins. For example:

> ERROR: This version of yosys includes the following plugins already: ghdl. Using plugins dynamically is not supported in this build. See reference_to_the_docs.

Ref open-tool-forge/fpga-toolchain#58 #1640

## Steps to reproduce the issue

- Download and extract [open-tool-forge/fpga-toolchain](https://github.com/open-tool-forge/fpga-toolchain).
- `PATH=/path/to/toolchain/bin:$PATH yosys -m ghdl`

whitequark, 2020-12-05:
This is quite annoying to implement nicely because `-m` takes a filename and tries several permutations of it, but you're using it like  it takes a plugin name. In my view, some refactoring is necessary first to clearly separate these two possible kinds of inputs. (Perhaps have one option that takes a full filename, and another that takes a plugin identifier?)

umarcor, 2020-12-05:
@whitequark I think I didn't explain myself properly. The proposal is not to discover and report the list of plugins that are available as modules (external shared libraries). This is about changing the error message when option `-m` is used in a Yosys built without plugin support. Attempting to use that option produces an error ("*This version of yosys is built without plugin support*"), which does not let the user know that some plugins/modules/frontends are built-in, even though external ones cannot be used.

I believe that the feature you were referring to is #1640.

PR #2512 (2021-01-01): plugin: enhance no-plugin error
Close #2408.
closed 2021-01-01 by whitequark