YosysHQ/yosys #2411: Unable to synthesize design in order to netlist to a json file

mickey4u, 2020-10-22:
I tried to synthesize the module below in order to obtain a json and then get the diagram from the json file. The system verilog code is shown below together with the diagram obtained
```
module hcd(input [1:7] icded,
        output logic [1:7] o_cded); 
        
 integer syndval ;
 function [2:0] synd (input [1:7] cw); 
    begin 
       synd[0] = cw[5] ^ cw[1] & cw[1] 
            ^ cw[3] ^ cw[7]; 
       synd[1] = cw[6] ^ cw[7] ^ cw[3]
            ^ cw[2]; 
       synd[2] = cw[5] ^ cw[4] ^ cw[6] 
            ^ cw[7]; 
     end 
 endfunction 
 
 always @(*) begin 
    o_cded = i_cded; 
    syndval = synd (i_cded); 
    if(syndval == 3'b0) begin end 
    else begin 
        o_cded[syndval] = ~i_cded[syndval]; 
    end 
 end 
endmodule
```
[image]

nakengelhardt, 2020-10-23:
You have a typo in the first line, so this is actually what your code describes! The input was declared as `icded` but the subsequent code uses the undeclared (and therefore by default single-bit) wire `i_cded`, and so a bunch of optimizations are applied based on the other bits being 0. If you change the input declaration to `i_cded`, the diagram will look more like you expect.

(By the way, yosys doesn't do any serious linting/error checking, as it's assumed that you will always run your code through something like verilator or iverilog first. In this specific instance, using `` `default_nettype none`` would also have helped.)

closed 2020-10-23 by nakengelhardt

mickey4u, 2020-10-23:
@nakengelhardt  Sorry it was a typo but corrected in what I tried on my computer.

nakengelhardt, 2020-10-23:
Can you tell me the full command you are running? Using `yosys -p "read_verilog -sv hcd.sv; hierarchy -top hcd; prep; show"` I get the following diagram:
[image]

mickey4u, 2020-10-23:
@nakengelhardt 
```
module hc_decoder(input [1:7] i_codeword,
        output logic [1:7] o_codeword); 
        
 integer synd_value ;
 function [2:0] syndrome (input [1:7] cw); 
    begin 
       syndrome[0] = cw[5] ^ cw[1] & cw[1] 
            ^ cw[3] ^ cw[7]; 
       syndrome[1] = cw[6] ^ cw[7] ^ cw[3]
            ^ cw[2]; 
       syndrome[2] = cw[5] ^ cw[4] ^ cw[6] 
            ^ cw[7]; 
     end 
 endfunction 
 
 always @(*) begin 
    o_codeword = i_codeword; 
    synd_value = syndrome (i_codeword); 
    if(syndrome_value == 3'b0) begin end 
    else begin 
        o_codeword[synd_value] = ~i_codeword[synd_value]; 
    end 
 end 
endmodule
```
```
yosys
read -sv dut.sv
synth -top hc_decoder
abc -g gates
write_json dut.json
```
after running `synth -top hc_decoder`
I get

```
=== hc_decoder ===

   Number of wires:                  3
   Number of wire bits:             15
   Number of public wires:           3
   Number of public wire bits:      15
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                  0

```

nakengelhardt, 2020-10-23:
You still have a typo on line 19, `syndrome_value` is undeclared.