YosysHQ/yosys #2415: File for ULX3S taking too long to build

rob-ng15, 2020-10-28:
Attempting to build this file for the ulx3s with yosys (current git and yosys-git-r10262.fd306b05) loops for a long time in the upper_sprites module, adding exclusive control bits, and flattening. The code for the de10nano compiles as expected very quickly in quartus.

Building via yosys takes nearly an hour, with quartus less than 10 minutes.

https://github.com/rob-ng15/Verilog-Playground/blob/master/j1eforth-verilog/build.v (extracted from https://github.com/rob-ng15/Silice-Playground/tree/master/j1eforth)

Until recently this was building at about the same speed with yosys as with quartus.

rob-ng15, 2020-10-30:
Using the latest git, file still hasn't completed in over 2 hours.

rob-ng15, 2020-10-31:
yosys command line

yosys -p 'scratchpad -copy abc9.script.flow3 abc9.script; synth_ecp5 -abc9 -json build.json' build.v

whitequark, 2020-10-31:
This could be related to memory inference. Try marking the memory in `upper_sprites` as `(*ram_block*)`.

rob-ng15, 2020-10-31:
I am relatively new to this, could you please explain? They are to my
knowledge marked as ram, but not 100% sure if exactly as you mean.

rob-ng15, 2020-10-31:
The other side to this is that the sprites are instantiated twice, once as lower_sprites, and once as upper_sprites. The lower _sprites does not get stuck.

Many thanks for responding.

mwkmwkmwk, 2020-10-31:
> This could be related to memory inference. Try marking the memory in `upper_sprites` as `(*ram_block*)`.

This is very unlikely, since it hangs in the `share` pass, long before any `memory_*` pass is run

rob-ng15, 2020-10-31:
@sylefeb has tried putting the ram block markjng, on and it made no
difference.

rob-ng15, 2020-11-01:
I have put (*ram_block*) on every intended ram block, still no difference, yosys keeps looping round in the same place. Can only conclude it is not a memory inference issue.

rob-ng15, 2020-11-05:
Curious if this is related to #2389 ?

rob-ng15, 2020-11-05:
Update build.v and yosys-output.txt https://github.com/rob-ng15/Silice-Playground/tree/master/YOSYS-ISSUE containes the build.v and yosys-output.txt ( in gzip )

rob-ng15, 2020-11-05:
With

`Yosys 0.9+3667 (git sha1 e7f36d01e, gcc 10.2.0 -march=x86-64 -mtune=generic -O2 -fno-plt -fPIC -Os)
`

Build is still going after 160 minutes, 4 times as long, with no sign of completion.

ZipCPU, 2020-11-05:
Let me just ask: can the ECP5 even support a dual clock memory with one write port on each clock?

    always @(posedge clock0) begin
      if (in_wenable0) begin
        buffer[in_addr0] <= in_wdata0;
      end else begin
        out_rdata0 <= buffer[in_addr0];
      end
    end
    
    always @(posedge clock1) begin
      if (in_wenable1) begin
        buffer[in_addr1] <= in_wdata1;
      end else begin
        out_rdata1 <= buffer[in_addr1];
      end
    end

Dan

rob-ng15, 2020-11-05:
I don't know the answer to that question. The design certainly works on the
built design, when it eventually builds that is.

@sylefeb may be able to advise on the clocks as he created the Silice
language which generates the verilog with the block ram primitives.

sylefeb, 2020-11-05:
Hi - I haven't been able to dive back into @rob-ng15 latest changes (will do as soon as I can). But in the previous versions, the dual clock memories were used with one port read-only and one port write-only, and yosys correctly inferred them. I have a number of other designs where this behaves properly as well (but - and I believe this was on ECP5 - only with this setup, otherwise logic is indeed used instead of BRAM, and this results in increased synthesis times).

@rob-ng15 if you have the opportunity, could you synthesize with a small number of sprites layers so that it terminates in reasonable time, and send me the yosys log? I'll verify the brams are properly inferred.

sylefeb, 2020-11-05:
I think @ZipCPU is correct - thanks for pointing us in the right direction - there is an issue with the bram contained in the sprite_layer module. In the log we can see:
```
Processing top.__main.lower_sprites.sprite_0.__mem__tiles.buffer:
  Properties: ports=2 bits=1024 rports=1 wports=1 dbits=16 abits=6 words=64
....
  No acceptable bram resources found.
```
repeated several times.
Am I correct this all gets turned into logic which then creates a heavy burden onto the optimizations?

@rob-ng15 it is important for these brams (and others) that the wenable0/1 is set (and stays) to either 0 or 1. I will make sure there is a check and warning about this in the future. Let's see if that resolves the problem!

rob-ng15, 2020-11-05:
I've gone through and replaced such things as tiles.wenable1 :=
writer_active; with tiles.wenable1 := 0; and flipped the write flag later
in the code. Is this what you meant?

sylefeb, 2020-11-05:
No, I meant that the write flag should always be 1 and never change; yosys will then be able to properly infer BRAMs (use tiles.wenable1 := 1, and always write the last value at the same address every clock).

rob-ng15, 2020-11-07:
Closing for now, as the issue seems to be resolved with the changes to the BRAM.

I've had to strip out some logic from the sprite layer code, otherwise the loop resumes.

rob-ng15, 2020-11-30:
Re-opening...

yosys-git-r10262.fd306b05 can compile my project in <20 minutes.

With yosys-git-0.9+3710.r10355.2116c5858 has so far been running for over an hour, two problems I noticed, whereby it waited for over 15 minutes in each of these stages:

`
2.32.3. Executing OPT_MUXTREE pass (detect dead branches in mux trees).                                                            
Running muxtree optimizer on module \top..                                                                                         
  Creating internal representation of mux trees.                                                                                   
  Evaluating internal representation of mux trees.                                                                                 
  Giving up (too many iterations)                                                                                                  
Removed 0 multiplexer ports.                                                                                                       
<suppressed ~100000 debug messages>                                                                                                

...

2.32.10. Executing OPT_MUXTREE pass (detect dead branches in mux trees).
Running muxtree optimizer on module \top..
  Creating internal representation of mux trees.
  Evaluating internal representation of mux trees.
  Giving up (too many iterations)
Removed 0 multiplexer ports.
<suppressed ~4961 debug messages>
`

rob-ng15, 2020-11-30:
It then sticks at `2.44.15.3. Executing XAIGER backend.` with 2% CPU usage.

Killed the build after 2 hours and no completion. What files are needed from me to assist with this issue?

For info, all BRAMS are instantiated correctly with yosys-git-r10262.fd306b05.

rob-ng15, 2020-12-06:
Closing, as yosys changed how block ram is inferred. Had to update code.

closed 2020-12-06 by rob-ng15