YosysHQ/yosys #2418: Sythesis ignores ambiguous contruct without error or warning.

sathibault, 2020-10-29:
## Steps to reproduce the issue

```
ghdl bug01.vhd
yosys -s bug01.s
```

## Expected behavior

The ambiguous code is:
```
  process (i70,i85,i58,i84,i53,i75)
  begin
    if (i85(1) = '1') then
      v_i82_i79_i80_i81 <= i70;
    elsif (i84(0) = '1') then
      v_i82_i79_i80_i81 <= i58;
    elsif (i84(1) = '1') then
      v_i82_i79_i80_i81 <= i53;
    elsif (i85(0) = '1') then
      v_i82_i79_i80_i81 <= i75;
    end if;
  end process;
```

The cases are actually exhaustive within the context of the design but not provable without knowledge about i84/i85.

I would expect the tools to either:
1. treat the code as exhaustive and implement (other synthesis tools do this)
2. report an error

## Actual behavior

The logic is complete discarded (outputs xxxx) and no error or warning is given.

whitequark, 2020-10-29:
I think this should be reported against GHDL, not Yosys.

Xiretza, 2020-10-29:
Also, without `bug01.s`, this can't be reproduced at all.

daveshah1, 2020-10-30:
Last I checked GHDL would error out on something like this on its own, as latches are not supported in ghdlsynth, so there must be something more to the code that you haven't shown that causes a mis-synthesis instead.

sathibault, 2020-10-30:
@whitequark I think you're right, I tried to reproduce this in Verilog and couldn't.

@Xiretza Not sure why the attachment didn't work the first time, but I've added it again.

@daveshah1 Nope that's the whole thing, see attached.

daveshah1, 2020-10-30:
This is definitely a GHDL issue.
```
ghdl --synth bug01.vhd -e bug01
```
```vhdl
library ieee;
use ieee.std_logic_1164.all;
entity bug01 is
  port (
    signal clk: in std_ulogic;
    signal i70: in std_ulogic_vector (23 downto 0);
    signal i58: in std_ulogic_vector (23 downto 0);
    signal i53: in std_ulogic_vector (23 downto 0);
    signal i75: in std_ulogic_vector (23 downto 0);
    signal i84: in std_ulogic_vector (1 downto 0);
    signal i85: in std_ulogic_vector (1 downto 0);
    signal i36_val: out std_ulogic_vector (23 downto 0)
  );
end entity;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

architecture rtl of bug01 is
  signal wrap_clk: std_logic;
  subtype typwrap_i70 is std_logic_vector (23 downto 0);
  signal wrap_i70: typwrap_i70;
  subtype typwrap_i58 is std_logic_vector (23 downto 0);
  signal wrap_i58: typwrap_i58;
  subtype typwrap_i53 is std_logic_vector (23 downto 0);
  signal wrap_i53: typwrap_i53;
  subtype typwrap_i75 is std_logic_vector (23 downto 0);
  signal wrap_i75: typwrap_i75;
  subtype typwrap_i84 is std_logic_vector (1 downto 0);
  signal wrap_i84: typwrap_i84;
  subtype typwrap_i85 is std_logic_vector (1 downto 0);
  signal wrap_i85: typwrap_i85;
  subtype typwrap_i36_val is std_logic_vector (23 downto 0);
  signal wrap_i36_val: typwrap_i36_val;
  signal v_i82_i79_i80_i81 : std_logic_vector (23 downto 0);
  signal n15_o : std_logic;
  signal n18_q : std_logic_vector (23 downto 0);
begin
  wrap_clk <= clk;
  wrap_i70 <= typwrap_i70(i70);
  wrap_i58 <= typwrap_i58(i58);
  wrap_i53 <= typwrap_i53(i53);
  wrap_i75 <= typwrap_i75(i75);
  wrap_i84 <= typwrap_i84(i84);
  wrap_i85 <= typwrap_i85(i85);
  i36_val <= std_ulogic_vector(wrap_i36_val);
  wrap_i36_val <= n18_q;
  -- bug01.vhd:17:10
  v_i82_i79_i80_i81 <= (23 downto 0 => 'X'); -- (signal)
  -- bug01.vhd:34:26
  n15_o <= '1' when rising_edge (wrap_clk) else '0';
  -- bug01.vhd:34:5
  process (wrap_clk)
  begin
    if rising_edge (wrap_clk) then
      n18_q <= v_i82_i79_i80_i81;
    end if;
  end process;
end rtl;
```

closed 2020-10-30 by daveshah1