YosysHQ/yosys #2419: Yosys reading the wrong execution path

PhilW92, 2020-10-29:
## Short description

The error occurs when installing yosys in a user defined directory. When executing yosys as a library (here: pyosys) passes, like techmap, that look for files inside `/share` folder can't find that folder since yosys reads the execution path from the environment (e.g. `/usr/bin`).

## Steps to reproduce the issue

Enable pyosys by adjusting the Makefile:
L21: ENABLE_LIBYOSYS := 1
L26: ENABLE_PYOSYS := 1

Install yosys in a non-standard destination by manually setting the `PREFIX` variable in Makefile (either add a `PREFIX =` line or use `make --prefix=**` when building)

In my case i used `~/.local` as the destination directory.

Set the installation directory of libyosys:
`L174: PYTHON_DESTDIR := $(shell $(PYTHON_EXECUTABLE) -c "import site; print(site.getsitepackages()[-1]);")`
to
`PYTHON_DESTDIR := $(PREFIX)/python`

Add yosys executable folder to PATH by adding `export PATH=~/.local/bin:$PATH` to `~/.bashrc`
Add libyosys folder by adding `export PYTHONPATH=~/.local/python:$PYTHONPATH` to `~/.bashrc`

Run the `techmap` pass inside a python script like
```
import libyosys as ys
design = ys.Design()
ys.run_pass("read_verilog example.v", design)
ys.run_pass("techmap", design)
```

The techmap pass internally calls the verilog frontend to load `techmap.v` file (found in `share`-directory).

The destination of the share directory is gathered at
https://github.com/YosysHQ/yosys/blob/e2a39bb1e71b97d423a3c2233abce996376e25dd/kernel/yosys.cc#L755
and adding `/share` to it.

## Expected behavior

Techmap pass is executed successfully since all paths are built correctly. The path to the file `techmap.v` is created adding `/share` to the execution path returned by the environment (here the execution path of python3: `/usr/bin`)

## Actual behavior

The pass ends with an error:
```
5. Executing TECHMAP pass (map to technology primitives).
ERROR: proc_share_dirname: unable to determine share/ directory!
```
The error is thrown by
https://github.com/YosysHQ/yosys/blob/e2a39bb1e71b97d423a3c2233abce996376e25dd/kernel/yosys.cc#L856

## Quick fix
A quick way to fix the issue is to change
`L761: return std::string(path, buflen);`
to
`return ABS_PATH_TO_BINFOLDER`

## Possible fix
Use the installation directory as gathered by the makefile but this prohibits the user from moving the created yosys directories after installation

## Discussion
This problem could also occur when running yosys as a library inside any c++ program that has a different execution path then the yosys intallation.

I do not have a usecase for that at hand so i can not test it.

I also discussed this with @btut

Xiretza, 2020-10-29:
Can't reproduce.

```
$ git rev-parse HEAD
e2a39bb1e71b97d423a3c2233abce996376e25dd
$ make clean
$ make -j8 PREFIX=$HOME/.local/yosys PYTHON_DESTDIR=$HOME/.local/python ENABLE_LIBYOSYS=1 ENABLE_PYOSYS=1 install
$ ~/.local/yosys/bin/yosys-config --datdir
[$HOME]/.local/yosys/share/yosys
$ env PATH=$HOME/.local/yosys/bin:$PATH PYTHONPATH=$HOME/.local/python:$PYTHONPATH python /tmp/yosystest.py
```

Runs successfully.

PhilW92, 2020-10-29:
That's interesting.

I tried your script and also could not reproduce the issue.

I then applied the destination changes to the Makefile by hand with:
```
PREFIX = ~/.local/test/yosys
PYTHON_DESTDIR := ~/.local/test/python
```
and the issue occured again

Xiretza, 2020-10-29:
Probably a bad idea to count on tilde expansion to occur everywhere in the Makefile - what does `yosys-config --datdir` output? Aside from the fact that patching the Makefile is unnecessary, using `$HOME` should be a lot safer.

PhilW92, 2020-10-29:
`yosys-config --datdir` outputs `~/.local/test/yosys/share/yosys` which is correct.

Using tilde obviously is not the best idea but changing it to $HOME does not solve the issue.

Xiretza, 2020-10-29:
It's actually not correct, yosys thinks the path starts with a literal tilde, which won't work (try `ls -al '~/.local/test/yosys/share/yosys'` to suppress tilde expansion in your shell).

PhilW92, 2020-10-29:
You are absolutely right, thanks for pointing that out. The problem still remains when changing Makefile lines to
```
PREFIX = $(HOME)/.local/test/yosys
PYTHON_DESTDIR := $(HOME)/.local/test/python
```

closed 2020-10-29 by PhilW92