YosysHQ/yosys #2421: Intermittent "failed to route" errors

EvgenyMuryshkin, 2020-10-30:
## Steps to reproduce the issue

I am getting failed to route error when building my project
I have created a test repo with all setup

1. Clone https://github.com/EvgenyMuryshkin/qusoc.git
2. Checkout branch **failed-to-route**
3. Open qusoc/tinyfpga in Atom with APIO (installed a week ago)
4. Run build

Build failed with "failed to route" error.

Changing line 22 in source file **top.v** from
```
  assign LED = counter[17];
```

to
```
  assign LED = counter[18];
```

fixes build.

## Expected behavior

Expect to build binaries

## Actual behavior

Route keeps failing for some configurations. Also happens when pulling out debug signals from the design etc.

daveshah1, 2020-10-30:
Are you using arachne-pnr by any chance (this was a known issue at the time), if so I would recommend using nextpnr. Either way this is not a Yosys issue, if nextpnr fails it should be reported on the nextpnr repo including the JSON netlist causing the route failure.

EvgenyMuryshkin, 2020-10-30:
I think it is running arachne-pnr. I will try to switch to nextpnr.

Thanks

EvgenyMuryshkin, 2020-10-31:
The latest nextpnr-ice40 seems to fix it.

Thanks.

closed 2020-10-31 by EvgenyMuryshkin