YosysHQ/yosys #2427: Yosys fails to read_verilog(generate RTLIL representation)

arun13e, 2020-11-02:
When running `read_verilog` on the attached file, yosys reports
_test.v:20: ERROR: Unsupported expression on dynamic range select on signal `$mem2bits$\wr_data_buf_reg$test.v:20$5'!_
This file is synthesizable in vivado. 

It is a poor attempt to reproduce this error, expect this verilog to be functionally incorrect.
[test.zip](https://github.com/YosysHQ/yosys/files/5477430/test.zip)

whitequark, 2020-12-05:
Minimized testcase:
```verilog
module test();
	reg [63:0] x;
	reg [63:0] y;
	reg [2:0] z[2:0];
	always @ (*)
		y = x[z[0]+:64];
endmodule
```

I can't understand how this part of genrtlil works, so I can't fix it.

whitequark, 2020-12-29:
Fixed in #2509.

closed 2020-12-29 by whitequark