YosysHQ/yosys #2429: No libreadline7 on Ubuntu 20

oloftangrot, 2020-11-04:
## Steps to reproduce the issue

I have upgraded a previous Ubuntu 18 machine where I had a working tool chain using yosys for ecp5 built locally
After the upgrade the tool chain ceased to work I have tried to rebuild yosys, prjtrellis and nextpnr.

But running yosys-abs fails with this error:
error while loading shared libraries: libreadline.so.7: cannot open shared object file: No such file or directory

And after some attempt to find out how this can be resolved it seems like there is not installation candidate for libreadline7 on
Ubuntu 20.

Nor can I find out why this shared library is used. The yosys-abs binary is rebuilt but somehow it depends on this non present library. It it possible to change to a version that is available like 6 or 8?

daveshah1, 2020-11-04:
Have you tried a full clean rebuild?

jakobwenzel, 2020-11-04:
I ran into the same issue on a system I upgraded from Ubuntu 18.04 to 20.04. I can confirm a full rebuild fixes this.

oloftangrot, 2020-11-08:
I did not if a clean build requires other methods than; make clean all;sudo make install. What is the correct method to make a clean build?

So my problem is that after my chosen method of a clean build the make processes still somehow uses libreadline7. I have not been able to reconfigure the build system to properly detect the installed version of libreadline. A 'configure'  is how I would have guessed it should work. Ok, to be fair I can see my self from the source code that GNU automake tools are not used so I hoped the version of readline could be manually defined.

oloftangrot, 2020-11-08:
@jakobwenzel Which version of readline does your build link in?

daveshah1, 2020-11-08:
Thinking about it, I don't think that `make clean` cleans abc which is the problem (if that's what you mean by 'yosys-abs'). Try doing `rm -rf abc` and then `make` which will freshly clone and build abc.

jakobwenzel, 2020-11-09:
@oloftangrot 
```shell
$ ldd yosys
	linux-vdso.so.1 (0x00007ffd5fd5b000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fec5116d000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fec5101e000)
	libreadline.so.8 => /lib/x86_64-linux-gnu/libreadline.so.8 (0x00007fec50fce000)
	libffi.so.7 => /lib/x86_64-linux-gnu/libffi.so.7 (0x00007fec50fc2000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fec50fbc000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fec50fa0000)
	libtcl8.6.so => /lib/x86_64-linux-gnu/libtcl8.6.so (0x00007fec50de2000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fec50dc7000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fec50bd5000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fec51ad3000)
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007fec50ba5000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fec50b82000)
```

@davesha1 IIRC `make clean` did not do it for me, so I used `git clean -dfx`.

oloftangrot, 2020-11-12:
gti clean -dfx did not work for me. The build still links to libreadline.so.7 which is not found.  I did a small test program that used and linked libreadline and the compiler, g++, apparently used libreadline.8.0. I don't claim that this provides evidence for the yosys build system to be faulty but at least my machine seems configured to provide a linkable readline library that the compiler is aware of.

oloftangrot, 2020-11-14:
rm -rf abc 
resolved the issue.

closed 2020-11-14 by oloftangrot

b224hisl, 2023-05-10:
Excuse me, I'm using yosys-abc inside OpenRoad-flow-script. I also meet such kind of problem: