YosysHQ/yosys #2431: CXXRTL RFE: Option to ignore $assert RTLIL cells

tomverbeure, 2020-11-07:
While experimenting with the Yosys ghdl plugin and CXXRTL, I ran into the issue of VHDL assert statements (like [this one](https://github.com/tomverbeure/cxxrtl_eval/blob/dcedc49661a13e1365eec68bd94a7d0b1b53d1f0/riscv_vhdl_cosim/ORCA-risc-v/instruction_fetch.vhd#L67), now commented out) being converted into RLTIL $assert comments.

CXXRTL doesn't support $assert cells and just bails.

It'd be nice for there to be a command line option in write_cxxrtl to just ignore all $assert cells.

Or is there a generic way in Yosys to remove all $assert cells from a design?

(Even better would be if it would convert the cell into a real check during simulation, but that's probably asking too much.)

daveshah1, 2020-11-07:
> Or is there a generic way in Yosys to remove all $assert cells from a design?

`chformal -remove` should do it

whitequark, 2020-11-07:
> (Even better would be if it would convert the cell into a real check during simulation, but that's probably asking too much.)

This is not only planned but also something I aim to implement soon.

whitequark, 2024-01-20:
Will be implemented by https://github.com/YosysHQ/yosys/pull/4128.

closed 2024-02-26 by whitequark