YosysHQ/yosys #2434: Add command or option to split assignments

pepijndevos, 2020-11-10:
Given the following example, `synth_gowin` produces code that cannot be parsed by the vendor toolchain.

```verilog
module top (input clk, output [1:0] leds);

reg [1:0] ctr;
always @(posedge clk)
	ctr <= ctr + 1'b1;

assign leds = ctr;

endmodule
```
Compile as follows. `-noiopads` is used for a minimal example, but the same behaviour can be seen _with_ iopads in more complex designs.
```
read_verilog blinky.v
synth_gowin -noiopads -vout vendorblinky.vg
dump
```
`synth_gowin` uses `splitnets` to split up nets in single wires as needed for the vendor netlist. However, it does not split up the connection. So the last line of the generated verilog will read as follows, which the vendor tool cannot parse.
```verilog
  assign leds = { \ctr[1] , \ctr[0]  };
```
The vendor tools gives
```
ERROR  (PA2000) : "/home/pepijn/code/apicula/vendorblinky.vg":47 | syntax error near token '{'
```
So a way is needed to split up the assignment itself. Note that the `-ports` option to `splitnets` does not help here, and only further complicates the LHS of the assignment.

I guess I'm kinda the de facto maintainer of `synth_gowin`, so I'm happy to take a stab at implementing this behaviour. I'm just hoping for some input from people more familiar with yosys internals on how to go about this.

Ravenslofty, 2020-11-12:
Isn't this just #1286 all over again?

pepijndevos, 2020-11-12:
You are completely right. Derp.

closed 2020-11-12 by pepijndevos