YosysHQ/yosys #2437: write_json does not write port directions

ti6wb, 2020-11-11:
Hi,
I want to use netlistsvg to visualize the synthesized circuit. Everything is fine, when I only synthesize the design, but if I map the design to a liberty based library, write_json does not write port direction.
It works with this synthesis script:

```
read_verilog counter.v; 
synth; 
write_json counter.json;
```

The cells in the generated json file consists of "port directions".

When I synthesize with:

```
read_verilog counter.v; 
synth ; 
dfflibmap -liberty cmos_cells.lib; 
abc -liberty cmos_cells.lib;
write_json counter.json;
```
then the port directions are missing and the generated .svg file shows the cells but no connections.

[counter.zip](https://github.com/YosysHQ/yosys/files/5524018/counter.zip)

daveshah1, 2020-11-11:
Doing `read_liberty -lib cmos_cells.lib` after read_verilog should load the blackboxes and thus port directions.

ti6wb, 2020-11-18:
This solves my problems. Thanks a lot.

closed 2020-11-18 by ti6wb