YosysHQ/yosys #2440: enums recognized in another file, but not typedef

norandomtechie, 2020-11-16:
I have two files: defs.vh and top.sv.

In defs.vh: 
`typedef enum logic [7:0] {TERMA=1, TERMB=2, TERMC=4, TERMD=8} term_t;`

In top.sv:
```
`default_nettype none
`include "defs.vh"
module fsm (input clk, reset, output logic [7:0] out);
  term_t foo;
  assign out = foo;
  always_ff @(posedge clk, posedge reset)
    if (reset)
      foo <= TERMA;
    else
      case(foo)
        TERMA: foo <= TERMB;
        TERMB: foo <= TERMC;
        TERMC: foo <= TERMD;
        TERMD: foo <= TERMA;
      endcase
endmodule
```

## Steps to reproduce the issue

Using the current Yosys git master, I ran the following command:
`yosys -p "read_verilog -sv defs.vh top.sv; synth_ice40 -top fsm; write_verilog struct_top.sv"`

## Expected behavior

The resulting structural Verilog should have implemented a circular FSM with the help of the term_t type.

## Actual behavior

A syntax error is thrown at the line "term_t foo" in top.sv: "top.sv:15: ERROR: syntax error, unexpected ';', expecting '(' or '['".  The error doesn't make much sense, but I assume it's because it doesn't recognize term_t, despite Yosys having parsed the defs header file first.

It works if I copy the typedef from defs.vh back into top.sv, but if we're trying to keep our definitions separate, then it doesn't work.  Is this a bug or am I doing something wrong here?

daveshah1, 2020-11-18:
I am seeing totally different behaviour here.

Doing
```
yosys -p "read_verilog -sv defs.vh top.sv; synth_ice40 -top fsm; write_verilog struct_top.sv"
```
gives me
```
1. Executing Verilog-2005 frontend: defs.vh
Parsing SystemVerilog input from `defs.vh' to AST representation.
Successfully finished Verilog frontend.

2. Executing Verilog-2005 frontend: top.sv
Parsing SystemVerilog input from `top.sv' to AST representation.
Generating RTLIL representation for module `\fsm'.
top.sv:3: ERROR: enum item \TERMA already exists
```

removing `defs.vh` from the command line (because it is included in top.sv anyway) and it works correctly

norandomtechie, 2020-11-20:
I accidentally mixed up the files I was testing when I posted this.  Sorry about that - the output you posted is what I'm getting.

I didn't realize that `include removes the need to specify the file to Yosys - thanks for pointing that out!  

Is the other way around (no include directive, but specifying the file) possible at all?  I tried it out by removing the `include directive from top.sv, and then ran the following:

`yosys -p "read_verilog -sv defs.vh top.sv; synth_ice40 -top fsm; write_verilog struct_top.sv"   # note the use of defs.vh`

which gives me this:

```
 Yosys 0.9+3675 (git sha1 71ca9a82, gcc 7.5.0-3ubuntu1~18.04 -fPIC -Os)

-- Running command `read_verilog -sv defs.vh top.sv; synth_ice40 -top fsm; write_verilog struct_top.sv' --

1. Executing Verilog-2005 frontend: defs.vh
Parsing SystemVerilog input from `defs.vh' to AST representation.
Successfully finished Verilog frontend.

2. Executing Verilog-2005 frontend: top.sv
Parsing SystemVerilog input from `top.sv' to AST representation.
top.sv:3: ERROR: syntax error, unexpected ';', expecting '(' or '['
```

Line 3 is where "term_t foo;" is.  Shouldn't it not throw an error in this case?  It's not really an issue for me, but it makes things easier for my use case for Yosys (receiving a list of files from a user, saving them to a folder, and passing the listing directly to Yosys to synthesize the given files).

whitequark, 2020-12-05:
> Is the other way around (no include directive, but specifying the file) possible at all?

I'm fairly certain that SystemVerilog does not specify that typedefs are preserved between files (and frankly it is weird enough already that preprocessor definitions do...)

udif, 2020-12-05:
I can add my 2 cents here.
From my own experience with C*****e tools (one of the "Big 3"), when you specify multiple files via a file list ("-f filelist.f"), the files are treated as if they were one giant file concatenated from all the files specified. This means ANYTHING that appears in an earlier file will be recognized by files down the list.
I haven't checked whether this behavior is defined in the spec or not.

whitequark, 2020-12-05:
That's surprising to me, but I agree that it is an argument in favor of supporting this behavior.

zachjs, 2021-02-06:
I agree the default behavior for most tools I've encountered is to compile all inputs under a single compilation unit. This concept is made explicit in the LRM, which states:

> The exact mechanism for defining which files constitute a compilation unit is tool-specific. However, compliant tools shall provide use models that allow both of the following cases:
> a) All files on a given compilation command line make a single compilation unit (in which case the declarations within those files are accessible following normal visibility rules throughout the entire set of files).
> b) Each file is a separate compilation unit (in which case the declarations in each compilation-unit scope are accessible only within its corresponding file).

My one concern is that carrying definitions across files could break some existing scripts if, for example, the same typedef exists in multiple files (likely as a workaround for the existing behavior).

PR #2671 (2021-03-17): sv: carry over global typedefs from previous files
This breaks the ability to use a global typename as a standard identifier in a subsequent input file. This is otherwise backwards compatible, including for sources which previously included conflicting typedefs in each input file.

I'm not sure why this logic is in `add_package_types`, but I figure I'd just fix it in place rather than making a larger change.

Fixes #2440
closed 2021-03-17 by zachjs