YosysHQ/yosys #2462: The 'design' command does not clear context as expected

christian-krieg, 2020-11-30:
I use the 'design' command to create several copies of the same design, applying different synthesis steps to each of these copies. Contrary to what I expect, the order matters in which I load/store these designs.

## Steps to reproduce the issue

I attach a simple design of a counter (counter.v). I do some initial processing (`proc`, `opt -full`), and then I save two copies of the design (`1` and `2`). Then, I load each design, and perform synthesis steps on each one of them. Other than expected, I get different synthesis results based on the order I perform the actions on the designs.

### Example design

This is the Verilog model of my example design (`counter.v`):
```
module counter (clk, rst, mode, cnt);

    input  wire clk;
    input  wire rst;
    input  wire mode;
    output reg signed [9:0] cnt = 10'sd42;

    reg signed [9:0] summand;

    always @* begin
        summand = 10'sd0;
        if (mode) begin
            if (cnt != -10'sd111) begin
                if (cnt <= 10'sd306) begin
                    summand = 10'sd4;
                end
            end else if (cnt <= 10'sd302) begin
                summand = 10'sd8;
            end
        // mode = 0
        end else begin
            if (cnt != -10'sd96) begin
                if (cnt >= -10'sd376 ) begin
                    summand = -10'sd11;
                end
            end else if (cnt >= -10'sd365) begin
                summand = -10'sd22;
            end
        end
    end

    always @(posedge clk) begin
        if (rst)
            cnt <= 10'sd42;
        else begin
            cnt <= cnt + summand;
        end
    end
endmodule
```

### Test case 1

When I call the commands in the following order, the `stat` command at the end of the script yields a cell count of `39` for `SB_LUT4` cells:
```
read_verilog counter.v
proc 
opt -full

design -save 1
design -save 2

design -load 1
synth_ice40
stat
design -save 1

design -load 2
techmap
opt -full
design -save 2

design -load 1
stat
```
Output of `stat`:
```
=== counter ===

   Number of wires:                 27
   Number of wire bits:            162
   Number of public wires:          27
   Number of public wire bits:     162
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                 89
     SB_CARRY                       40
     SB_DFFSR                       10
     SB_LUT4                        39
```

### Test case 2

When I swap the actions for the design copies `1` and `2`, `stat` yields `33` instances of `SB_LUT4` cells.

```
read_verilog counter.v
proc 
opt -full

design -save 1
design -save 2

design -load 2
techmap
opt -full
design -save 2

design -load 1
synth_ice40
stat
design -save 1

design -load 1
stat
```
Output of `stat`:
```
=== counter ===

   Number of wires:                 29
   Number of wire bits:            169
   Number of public wires:          29
   Number of public wire bits:     169
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                 83
     SB_CARRY                       40
     SB_DFFSR                       10
     SB_LUT4                        33
```

## Expected behavior

The `stat` command outputs the same number of cell counts independently of the order in which the designs are stored/loaded.

## Actual behavior

The `stat` command outputs **different** cell counts, so it appears to depend on the order in which the designs are stored/loaded.

## File attachments

I packaged the above code to reproduce the error. The zip archive contains `counter.v`, `test-case-1.ys`, and `test-case-2.ys`. After unzipping, the following commands reproduce the error.

```
yosys -s test-case-1.ys
yosys -s test-case-2.ys
```
[mcve.zip](https://github.com/YosysHQ/yosys/files/5617899/mcve.zip)

mwkmwkmwk, 2020-12-05:
For the record: while we do consider it a bug, it's unlikely to be fixed any time soon, due to how yosys works.

The likely source of the problem here is IdStrings and iteration order of containers involving those; given that the set of IdStrings is global (and changing that is near-impossible), creating some IdStrings in the first pass will affect the iteration order in the second pass and mess with it.

whitequark, 2020-12-05:
Wait, I thought dicts are iterated in insertion order?

christian-krieg, 2020-12-06:
Thank you very much @mwkmwkmwk and @whitequark for analyzing the issue!

To make sure I understand correctly: You state that actual IdStrings influence the order in which a pass iterates over RTLIL objects, therefore creating different synthesis results for test cases 1 and 2?

I understand that it would be a substantial architectural change for Yosys to make RTLIL objects local to a design; but theoretically, would that solve the issue?

I guess a hint in the `design` help text would help users to make them aware of this non-intuitive behavior (just in case a fix won't be able soon). I can draft a line or two, if desired :)

daveshah1, 2020-12-06:
nextpnr implements context-specific IdStrings - mainly as a more performant alternative to the refcount and GC approach that Yosys takes - but tbh it is quite a nuisance because anything that needs to turn an IdString to/from a string also needs a pointer to the current context.

KrystalDelusion, 2025-05-06:
Given that `synth_ice40` uses `abc`, I think this is another case of #3713.  Breaking out the `map_luts` step of and adding a `rename -scramble-name` before the call to abc can also change the number of cells/wires in the final result, whether you run 1 or 2 first.

widlarizer, 2026-11-04:
As of #6391 autoidx is part of Design and this is fixed

closed 2026-11-04 by widlarizer