YosysHQ/yosys #2464: I have some question about some code in opt_clean.cc

yongshiwo, 2020-12-01:
## Steps to reproduce the issue

*Provide instructions for reproducing the issue. Make sure to include
all necessary source files. (You can simply drag&drop a .zip file into
the issue editor.)*

Also, make sure that the issue is actually reproducable in current git
master of Yosys.

See https://stackoverflow.com/help/mcve for some information on how to
create a Minimal, Complete, and Verifiable example (MCVE).

Please do not waste our time with issues that lack sufficient information
to reproduce the issue easily. We will simply close those issues.

Contact https://www.symbioticeda.com/ if you need commercial support for Yosys.

## Expected behavior

*Please describe the behavior you would have expected from the tool.*

## Actual behavior

*Please describe how the behavior you see differs from the expected behavior.*

This is code on line 330-336 in pass/opt/opt_clean.cc.
		for (auto &it2 : cell->connections_) {
			assign_map.apply(it2.second);
			raw_used_signals.add(it2.second);
			used_signals.add(it2.second);
			if (!ct_all.cell_output(cell->type, it2.first))
				used_signals_nodrivers.add(it2.second);
		}
I have a question the next two line.
			assign_map.apply(it2.second);
			raw_used_signals.add(it2.second);
			used_signals.add(it2.second);
I think it's right to adjust the order of  "assign_map.apply(it2.second);" and "raw_used_signals.add(it2.second);" 
                        raw_used_signals.add(it2.second);
			assign_map.apply(it2.second);
			used_signals.add(it2.second);

whitequark, 2020-12-05:
Please explain why you think the order should be changed.

zachjs, 2021-03-28:
I don't see an obvious reason why this would affect the behavior of the pass. I am closing this due to inactivity.

closed 2021-03-28 by zachjs