YosysHQ/yosys #2465: SMT2 Backend Treats Non-POT-Sized Memories as next higher POT

jakobwenzel, 2020-12-01:
## Steps to reproduce the issue

This is a simplified version of `syncram_9_4.v` from https://github.com/YosysHQ/yosys-bench/tree/master/verilog/benchmarks_small/ram:

```
module syncram_9_4 (
    clk,
    addr_in,
    data_out    
);

input clk;
input [8:0] addr_in;
output reg [3:0] data_out;

reg [3:0] mem [0:8];

always @(posedge clk)
begin
  data_out <= mem[addr_in];
end

endmodule
```

Reading, creating a Miter circuit comparing two instances of this module and writing to SMT2 with this Yosys script:

```
read_verilog syncram_9_4.v
miter -equiv -make_assert syncram_9_4 syncram_9_4 miter
prep -top miter
setundef -zero -init -undriven -params
write_smt2 miter.smt2
```
Note that `setundef` adds zero-initialization to all memory lines.

Now running through `yosys-smtbmc`:

```
yosys-smtbmc -s boolector --dump-vlogtb tb.v miter.smt2
```

For some reason, `yices` does not like this smt2 file (`Unexpected EOF response from solver.`), so I am using boolector. If I am doing anything wrong, please tell me :)

## Expected behavior

The assertions should pass.

## Actual behavior

```
##   0:00:00  BMC failed!
##   0:00:00  Assert failed in miter: $auto$miter.cc:233:create_miter_equiv$5
##   0:00:00  Status: FAILED (!)
```

## Analysis
When writing the memory to SMT2, its actual size is never written, only its number of address bits. This can be seen when changing the memory's size in the original source, e.g. to `reg [4-1:0] mem [0:9];`. The diff between the two smt2 files is:

```
--- miter_9.smt2	2020-12-01 16:07:26.124944308 +0100
+++ miter_10.smt2	2020-12-01 16:07:39.286427423 +0100
@@ -32,6 +32,7 @@
   (= (bvand (select (|syncram_9_4#3#0| state) #b0110) #b1111) #b0000) ; mem[6]
   (= (bvand (select (|syncram_9_4#3#0| state) #b0111) #b1111) #b0000) ; mem[7]
   (= (bvand (select (|syncram_9_4#3#0| state) #b1000) #b1111) #b0000) ; mem[8]
+  (= (bvand (select (|syncram_9_4#3#0| state) #b1001) #b1111) #b0000) ; mem[9]
 ))
 (define-fun |syncram_9_4_h| ((state |syncram_9_4_s|)) Bool true)
 (define-fun |syncram_9_4_t| ((state |syncram_9_4_s|) (next_state |syncram_9_4_s|)) Bool (and
```
The added line is due to `setundef` initializing the new memory location. As both a 9-entry and a 10-entry memory have an address width of 4 bits, the actual memory did not change.

The memory's size is therefore rounded up to the next power of two. It no longer has 9 lines, but 16. As these lines do not exist in the design, they are not initialized by `setundef`.  The solver is free to choose any initialization for them, as can be seen in the generated testbench:
```
    UUT.gate.mem[4'b1010] = 4'b0111;
    UUT.gold.mem[4'b1010] = 4'b1000;
```
As the solver was free to choose different initializations, the assertion failed.

georgerennie, 2025-06-07:
So theres two parts to this, the first is that its valid for your example to not compare equivalent - for `addr_in` greater than 9 `data_out` is `x`, and in standard formal flows different instances of `x`s (e.g. from the two sides of the miter) are not required to compare equal, they are treated as unconstrained free variables. See https://github.com/YosysHQ/sby/issues/324 for a similar issue, one way of solving this is adding `assume(addr_in <= 9)`.

Despite that, there is an issue with how the smt2 backend sizes non pow2 memories up to pow2. See the following example:
```verilog
module top (
	input clk
);

logic [3:0] mem [0:8];
logic write;
logic [8:0] addr_in, addr_out;
logic [3:0] data_in;
wire [3:0] data_out = mem[addr_out];

always @(posedge clk) begin
	if (write)
		mem[addr_in] <= data_in;
end

logic [7:0] cycle = 0;
always_ff @(posedge clk) begin
	if (cycle != '1)
		cycle <= cycle + 1;
end

always_comb begin
	write = '0;
	data_in = '0;
	addr_in = '0;
	addr_out = '0;
	case (cycle)
		0: begin
			write = 1;
			addr_in = 12;
			data_in = 15;
		end
		1: begin
			addr_out = 12;
			assert(data_out == 15);
		end
		2: begin
			addr_out = 13;
			assert(data_out == 15);
		end
	endcase
end

endmodule
```

In this example, we write the value 15 to address 12 (outside the bounds of the memory). All further reads of address 12 should read `x`, as those memory addresses do not actually exist, however in practice for the smt2 and btor2 backends they are being modelled as actual memory addresses so further reads always read 15. Thus when we try reading address 12, the assertion that it is equal to 15 passes, whereas when we try reading address 13 (which has had no value written yet) the assertion fails as this is not required to be equal to 15.

This behaviour is not present in the AIG backend, and I think needs to be fixed by modelling reads as `in_bounds(read_addr) ? mem[read_addr] : 'x` in smt2 and btor2 backends

georgerennie, 2025-06-07:
Actually AIG backend also seems to be doing something bad... just instead of modelling these addresses as memories it seems to be modelling them as always reading 0