YosysHQ/yosys #2472: 'std::out_of_range' at 'dict::at()` when not all SB_LUT4 inputs are used

Disasm, 2020-12-03:
## Steps to reproduce the issue

Run `yosys -p "synth_ice40 -json top.json" top.v` with this file:

```verilog
module top(
    input wire in,
    output wire out
);
SB_LUT4 lut_inst(
    .O(out),
    .I3(in)
);
defparam lut_inst.LUT_INIT=16'hff00;
endmodule
```

## Expected behavior

Yosys produces an output with all unconnected SB_LUT4 inputs wired to zero.
From Lattice iCE Technology Library V3.0 document, `SB_LUT4` description:
> **Default Signal Values**
The iCEcube2 software assigns logic value ‘0’ to unconnected input ports.

## Actual behavior

`std::out_of_range` is raised during the `ICE40_WRAPCARRY` pass.

whitequark, 2020-12-05:
Should be fixed by https://github.com/YosysHQ/yosys/pull/2474.

mmicko, 2021-11-17:
Has been fixed in #2669.

closed 2021-11-17 by mmicko