YosysHQ/yosys #2482: Erronously instantiated SB_CARRY leads to crash

bnossum, 2020-12-09:
### Description
An error in the instantiation of `SB_CARRY` (input `I1` replicated) does not exit particularly gracefully. The last line in the log is:
    
    2.30. Executing ICE40_WRAPCARRY pass (wrap carries).

   
### Version
    Yosys 0.9+3652 (git sha1 623526d, clang 3.8.0-2ubuntu4 -fPIC -Os)

### Code to replicate this
      
    module Not_a_graceful_failure
      (input  a,b,
       output cy
       );   
       SB_CARRY l_cy0( .CO(cy),.CI(1'b0),.I1(b),.I1(a));   
    endmodule

### Result
    yosys -o obj_dir/top.json -p 'synth_ice40 -json obj_dir/R2008.json'  carryerr.v -g > obj_dir/synth.log
    terminate called after throwing an instance of 'std::out_of_range'
    what():  dict::at()
    Aborted (core dumped)

daveshah1, 2020-12-09:
Luckily, GH can transfer issues between repos these days (I've been looking for an excuse to try that out...)

bnossum, 2020-12-09:
Sorry for the wrong location, and thanks for this transfer.

commit 08510d72 (2020-12-09): Merge pull request #2483 from YosysHQ/pmgen_nice_error

Return nice error in pmgen generated code, fixes #2482
commit 82dcf78c (2020-12-09): Return nice error in pmgen generated code, fixes #2482
closed 2020-12-09 by mmicko