YosysHQ/yosys #2484: Question: "Don't Optimize" attribute in Yosys

jay20162016, 2020-12-09:
Is there a attribute that tells yosys to not optimize this?

I have code for a delay and it's optimizing it away, removing the delay. This delay is essential to my module. Thanks!

jay20162016, 2020-12-09:
also, `(* keep *)` or `(* dont_touch *)` doesn't work.

daveshah1, 2020-12-09:
`(* keep *)` is pretty much the only attribute Yosys has for this. If it isn't working then you'll need to provide a bit more info.

jay20162016, 2020-12-09:
I have this as my delay module:
```verilog
module delay #(
  parameter T = 3)
  (
  input d,
  output out);

  (* keep = 32'd1 *)
  wire [T:0] path;

  assign path[0] = d;
  assign out = path[T];

  genvar i;
  generate // optional if > *-2001
    for (i = 0; i < T; i = i + 1) begin
      (* keep *) (* dont_touch *)
      \$lut #(
          .WIDTH (4),
          .LUT (32'b1111_1111_0000_0000)
        ) lut (
          .A({path[i], 1'b0, 1'b0, 1'b0}),
          .Y(path[i+1])
        );
    end
  endgenerate

endmodule
```

after running `synth_ice40`, it turns into 
```verilog
(* dynports =  1  *)
(* top =  1  *)
(* src = "delay.v:1.1-38.10" *)
module delay(d, out);
  (* src = "delay.v:4.9-4.10" *)
  input d;
  (* src = "delay.v:5.10-5.13" *)
  output out;
  (* keep = 32'd1 *)
  (* src = "delay.v:8.14-8.18" *)
  wire [3:0] path;
  assign out = d;
  assign path = { d, d, d, d };
endmodule
```

daveshah1, 2020-12-09:
I would recommend using SB_LUT4 here rather than $lut

jay20162016, 2020-12-09:
Well, I'd like to make the design generic (not vendor-specific). Would there be a alternative that does that?

whitequark, 2020-12-10:
@jay20162016 Your module is inherently vendor-specific, so no.

jay20162016, 2020-12-10:
LUTs are vendor specific?

whitequark, 2020-12-10:
They are, of course. There are several issues relevant to your use case:
  * The exact delay you will get depends on the LUT input you use.
  * The `$lut` cell is treated like a multiplexer by the toolchain, and participates in LUT synthesis just like any other combinatorial cell. If you want a LUT primitive, you have to use a LUT primitive.

jay20162016, 2020-12-10:
> The exact delay you will get depends on the LUT input you use.

All I need is relative delay: A will complete after B.

> The $lut cell is treated like a multiplexer by the toolchain, and participates in LUT synthesis just like any other combinatorial cell. If you want a LUT primitive, you have to use a LUT primitive, and those are vendor-specific.

Is there then a way to prevent it from being optimized?

whitequark, 2020-12-10:
> Is there then a way to prevent it from being optimized?

Yes. Use a vendor-specific LUT primitive instead.

jay20162016, 2020-12-11:
Ok, thanks!

closed 2020-12-11 by jay20162016