YosysHQ/yosys #2486: BlockRAM inference on ecp5 fails even for simple example

MartinGeisse, 2020-12-13:
## Steps to reproduce the issue

The following code fails to infer a block RAM:

```
`default_nettype none
`timescale 1ns / 1ps

module MemoryTest(
	clk,
	address,
	writeEnable,
	writeData,
	readData
);

input clk;
input[10:0] address;
input writeEnable;
input[7:0] writeData;
output[7:0] readData;

reg[7:0] readDataRegister;
reg [7:0] memory [2047:0];

initial begin
	readDataRegister <= 8'h00;
end
always @(posedge clk) begin
	if (writeEnable) begin
		memory[address] <= writeData;
	end
	readDataRegister <= memory[address];
end
assign readData = readDataRegister;

endmodule
```

command: `yosys -p 'synth_ecp5 -json test.json' test.v`

## Expected behavior

This should result in a block RAM being inferred.

## Actual behavior

Lots of $__TRELLIS_DPR16X4 get inferred, which is probably distributed RAM.

mwkmwkmwk, 2020-12-13:
The problem here is that `readDataRegister` is initialized, and yosys currently doesn't support merging initializable read registers into memory.

This is on the memory inference rewrite TODOlist and should be fixed in a month or two. For now you can make it infer the blockRAM by removing the initial block.

MartinGeisse, 2020-12-13:
Confirmed, removing the initial block fixes the issue. Thank you for the quick reply!

closed 2020-12-13 by MartinGeisse