YosysHQ/yosys #2488: bug in 'opt'

alanminko, 2020-12-14:
The design "test3_3.v" attached below is correctly elaborated using "proc" but becomes logically incorrect after running "opt" using the latest version of Yosys.  The same result was produced by a two-year-old version of Yosys:

yosys> design -reset
yosys> read_verilog test3_3.v
yosys> proc
yosys> write_verilog -noattr test3_3_proc.v
yosys> opt
yosys> write_verilog -noattr test3_3_proc_opt.v

[yosys_bug.zip](https://github.com/YosysHQ/yosys/files/5686294/yosys_bug.zip)

daveshah1, 2020-12-14:
The problem is that without a call to `hierarchy` before `proc`, the parameterised module is not elaborated correctly, so downstream optimisation passes are incorrectly optimising based on the default parameterisation - and hence a port width of 8.

While this is definitely surprising behaviour and I think could be improved, always running `hierarchy` first is important for correct results.

modwizcode, 2021-01-06:
I've encountered a similar bug as well, I've included my test case which does include the hierarchy pass being run. 

It put the memory writes into a `process @(negedge 1'hx)` block. Strangely from what I can see if I were to dump with `write_rtlil` or export for cxxrtl with `write_cxxrtl` both produce functional results. 

Run make to regenerate the file. [testcase.zip](https://github.com/YosysHQ/yosys/files/5777753/testcase.zip)

closed 2021-08-13 by mmicko