YosysHQ/yosys #2493: Cornercase. Verilog frontend exits with assert instead of an error message

bnossum, 2020-12-17:
## Steps to reproduce the issue
Using `Yosys 0.9+3760 (git sha1 871fc34, clang 3.8.0-2ubuntu4 -fPIC -Os)`

    module top  # ( parameter AUTOPLACE = 0 )
       (
        input  rxck270,
        input  ck27start,
        output ubufclk135
        );
       
       generate
          if ( AUTOPLACE ) begin
             //             
          end else begin
             
             wire cmb_ck135 = ck27start ? 1'b1 : ~ubufclk135;           
             (* BEL="X1/Y11/lc3" *) SB_DFF reg_clk135(.Q(ubufclk135),.C(rxck270),.D(cmb_ck135));
    
             assign ubufclk135 = reg_clk135;   // This illegal statement trigger assert error.
             
          end
       endgenerate
    endmodule

## Expected behavior
An error message of some kind, on the illegal statement, for instance:

    top.v:20: ERROR: Failed to detect width for identifier \reg_clk135!

## Actual behavior

    ERROR: Assert `count_id(wire->name) == 0' failed in kernel/rtlil.cc:1801

whitequark, 2021-01-01:
@zachjs Could you take a look at this issue please?

zachjs, 2021-01-02:
In general, items in _unnamed_ generate blocks are simply inserted at the top level scope once that generate block is elaborated. The cell `reg_clk135` in this example is added to the top level module, but is never added to `current_scope`, as cells (and several other item types) are only added to the scope in the first pass of the top-level module items. I believe this is done to avoid enforcing that items are declared before their first use, as some tools allow this behavior despite it not being standards-compliant. Because `reg_clk135` is not seen (despite now existing at the top level), an auto wire to be created, leading to the assertion failure in genrtlil.

While I could add an error check to hide this, I believe the crucial issue here is the handling of unnamed generate blocks. Items in such blocks _should_ be added to the current scope, but should be prefixed with an autogenerated name. For example, the following should raise an error, but currently does not:
```SystemVerilog
`default_nettype none
module top1;
    wire x;
    generate
        if (1) wire y;
    endgenerate
    assign x = y;
endmodule
```
I believe this issue with unnamed generate blocks is also the root cause of #656.

closed 2021-02-04 by whitequark