YosysHQ/yosys #2500: CXXRTL: latest version with -g4 doesn't store all signals in vcd file

tomverbeure, 2020-12-22:
I'm my standard cxxrtl_eval project with latest version of Yosys and written a model with `write_cxxrtl -O6 -g4`.

When doing that, the following signal doesn't show up in the VCD file:
`bridge_write_streamUnbuffered_queueWithOccupancy_io_availability[7:0]`

It shows up fine in a Verilator generated VCD file.

(If necessary, I'll update this issue later using the normal template, but I wanted to file this before doing something else.)

Tom

whitequark, 2020-12-22:
That isn't supposed to happen! Please attach the IR and I'll investigate.

tomverbeure, 2020-12-23:
.ilang and .cpp file
[ExampleTop.sim.Og.zip](https://github.com/YosysHQ/yosys/files/5732675/ExampleTop.sim.Og.zip)

tomverbeure, 2020-12-29:
Here is a much smaller case that should be easier to debug: [testcase.zip](https://github.com/YosysHQ/yosys/files/5749628/testcase.zip)

```verilog
  always @(*) begin
    case(cur_state)
      `FsmState_binary_sequential_Idle : cur_state_string = "Idle     ";
      `FsmState_binary_sequential_FetchCoef : cur_state_string = "FetchCoef";
      `FsmState_binary_sequential_FetchData : cur_state_string = "FetchData";
      default : cur_state_string = "?????????";
    endcase
  end
```

Just like in the earlier test case, `cur_state_string` is a signal that's only assigned, and never used. It is present in the ilang, but not in the cpp file.

Yosys script:
```
read_verilog -formal FirEngine.v
hierarchy -check -top FirEngine
write_cxxrtl -O6 -g4 FirEngine.cpp
write_ilang FirEngine.cpp.ilang
```

whitequark, 2020-12-29:
Ah yes! That was *sort of* intended as a change, but you are right that these wires should be outlined. Thanks for minimizing the testcase!

By the way, are you aware of the enum-related attributes? I think I can make CXXRTL emit enums directly so you wouldn't need this hack. I currently don't have a complex netlist with enums, so if you do, and you'd like to see enums supported, I could use one.

tomverbeure, 2020-12-29:
> By the way, are you aware of the enum-related attributes?

I'm not, but this hack is not target specific and works for all Verilog simulator/waveform viewers that I know. It gets auto-generated in SpinalHDL.

whitequark, 2020-12-29:
> I'm not, but this hack is not target specific and works for all Verilog simulator/waveform viewers that I know.

It also hurts performance, but yes.

> It gets auto-generated in SpinalHDL.

Can SpinalHDL emit SystemVerilog enums? Those will get hooked up appropriately as well.

tomverbeure, 2020-12-29:
> Can SpinalHDL emit SystemVerilog enums? 

It does not.

PR #2873 (2021-07-16): cxxrtl: emit debug items for unused public wires
This greatly improves debug information coverage.

Fixes #2500.
commit 09218896 (2021-07-16): cxxrtl: emit debug items for unused public wires.

This greatly improves debug information coverage.

Fixes #2500.
closed 2021-07-16 by whitequark