YosysHQ/yosys #2505: Build on MSYS2 is broken

umarcor, 2020-12-26:
## Steps to reproduce the issue

Building Yosys on MSYS2 through a PKGBUILD recipe. See last commit in https://github.com/umarcor/yosys/commits/ci/msys2.

## Expected behavior

Yosys is successfully built on either MINGW64 or MINGW32.

## Actual behavior

On MINGW32, the failure is the following:

https://github.com/umarcor/yosys/runs/1609838330?check_suite_focus=true#step:5:88
```
/bin/sh: x86_64-w64-mingw32-g++: command not found
make: *** [Makefile:700: kernel/driver.o] Error 127
```

On MINGW64, the build was succesful until https://github.com/YosysHQ/yosys/commit/e825cf9d73a2126d518b2dc8f78114850c84ad32. See https://github.com/umarcor/yosys/actions/runs/444767078. It seems that https://github.com/YosysHQ/yosys/commit/b2221c1077d43a193c1a66d8017a7ed9ed1a2373 broke it: https://github.com/umarcor/yosys/runs/1609948936?check_suite_focus=true.

whitequark, 2020-12-26:
That doesn't seem right--the commit you mention does not change the build system.

umarcor, 2020-12-26:
It seems to be a syntax error, not related to the build system. In fact, the previous commit does work. Please, see https://github.com/umarcor/yosys/runs/1609948936?check_suite_focus=true#step:5:1236

whitequark, 2020-12-26:
Ah, yes. On Windows, it will be necessary to `#undef CONST` so that it would stop interfering with our code.

umarcor, 2020-12-26:
Thanks! That allowed to build 'master' on MINGW64 again!

What about MINGW32? Do you know whether Yosys is expected to work on Windows 32 bits? Was it ever tested/supported?

whitequark, 2020-12-26:
> What about MINGW32? Do you know whether Yosys is expected to work on Windows 32 bits? Was it ever tested/supported?

I don't know, but I think that if you fix the build system issue, it should compile and run fine.

umarcor, 2020-12-26:
My point is that `make config-msys2-64` exists, but `make config-msys2-32` does not. The build system tries using `x86_64-w64-mingw32-g++`, regardless of being MINGW64 (correct) or MINGW32 (incorrect). See https://github.com/umarcor/yosys/runs/1612345840?check_suite_focus=true#step:5:81.

With "fix the build system issue" you mean implementing `config-msys2-32` myself?

whitequark, 2020-12-26:
> With "fix the build system issue" you mean implementing `config-msys2-32` myself?

Yes.

PR #2511 (2020-12-28): Update MSYS2 build system
Close #2505
Close #2507

This PR is based on #2507.

CONFIG option `msys2` is renamed to `msys2-32`, for consistency. Furthermore, `PREFIX := $(MINGW_PREFIX)` is added to both `msys2-32` and `msys2-64`.
closed 2020-12-28 by whitequark