YosysHQ/yosys #2513: Simulation of empty ports does default to the correct size, errors and does not specify the net.

terriblefire, 2020-12-28:
## Steps to reproduce the issue

Attempt to **simulate** a blackbox simulation of a SB_PLL40_CORE in verilog and specify a port blank. i.e.

SB_PLL40_CORE pll_inst(.REFERENCECLK(CLKXTAL),
                              .PLLOUTCORE(PCLK),
                              .PLLOUTGLOBAL(CLKCPU),
                              .EXTFEEDBACK(),
                              .DYNAMICDELAY(),
                              .RESETB(1'b1),
                              .BYPASS(1'b0),
                              .LATCHINPUTVALUE(),
                              .LOCK(RESET),
                              .SDI(),
                              .SDO(),
                              .SCLK()
);

defparam pll_inst.DIVR = 4'b0000;
defparam pll_inst.DIVF = 7'b0011111;
defparam pll_inst.DIVQ = 3'b100;
defparam pll_inst.FILTER_RANGE = 3'b010;
defparam pll_inst.FEEDBACK_PATH = "SIMPLE";
defparam pll_inst.DELAY_ADJUSTMENT_MODE_FEEDBACK = "FIXED";
defparam pll_inst.FDA_FEEDBACK = 4'b0000;
defparam pll_inst.DELAY_ADJUSTMENT_MODE_RELATIVE = "FIXED";
defparam pll_inst.FDA_RELATIVE = 4'b0000;
defparam pll_inst.SHIFTREG_DIV_MODE = 2'b00;
defparam pll_inst.PLLOUT_SELECT = "GENCLK";
defparam pll_inst.ENABLE_ICEGATE = 1'b0;

This is perfectly valid verilog and synths fine.

[pll.zip](https://github.com/YosysHQ/yosys/files/5747737/pll.zip)

Run the simulate.sh script. 

Error 

ERROR: Assert `GetSize(sig) <= GetSize(value)' failed in passes/sat/sim.cc:240.

## Expected behaviour

Either specify the net that has the error or default to the correct net.

ERROR: <Insert net name here> Assert `GetSize(sig) <= GetSize(value)' failed in passes/sat/sim.cc:240.

## Actual behavior

Fails with a cryptic error. 

ERROR: Assert `GetSize(sig) <= GetSize(value)' failed in passes/sat/sim.cc:240.

daveshah1, 2020-12-28:
There's definitely a bug here; but as Yosys doesn't include a PLL sim model (and the built-in sim command is essentially limited to the synthesisable subset of Verilog and would struggle to model a PLL) then even with the bug fixed I wouldn't expect the example to 'work'.

terriblefire, 2020-12-28:
Its not about the example. If yosys gave better errors I could have worked out the sim wouldnt work in 10 mins instead of 3 hours. 

My assumption was that there was a compact model for the PLL.. otherwise why have a sim header? but that doesnt matter as much as making error reporting better.

whitequark, 2020-12-28:
> otherwise why have a sim header?

AFAIK, simulation cells are also used during synthesis (or rather, their interfaces are).

whitequark, 2020-12-28:
Sorry, closed by accident.

PR #2621 (2021-03-08): sim: Avoid a crash on empty cell connection.
Fixes #2513.
commit c00a2929 (2021-03-08): sim: Avoid a crash on empty cell connection.

Fixes #2513.
closed 2021-03-08 by mwkmwkmwk