YosysHQ/yosys #2516: CXXRTL: Strange timings and mismatch in dumped VCD

modwizcode, 2020-12-30:
I've been working with a basic simulation only fifo design that I plan to put into an ethernet design for simulation.

cxxrtl is being used to test the behavior of the fifo, however I've noticed that it seems like some of my signals when I used the built-in VCD functionality aren't having the correct timings. 

It seems like the `state` vector in the module `stupid_fifo` isn't correctly matching the assigned values of the concatenation of `wrote_first_word`, `started_up`, and `going_done`. Instead there's some sort of unexpected half cycle delay.

[image]

This could be entirely due to a misunderstanding. I rewrote most of my test environment as a top level testbench in verilog to iron out the timing of the driving signals, which I was having trouble understanding under cxxrtl. 

Moving them into a top level module made it possible to know that the behavior would match verilog signals driven along a rising edge of the clock, which seemed unclear to do in cxxrtl. The correspondence between the cout prints of the signals and the dumped values in the vcd also don't seem to match up.

I'm attaching a copy of the files for the project, you should be able to just do a make and test it out.

[test.zip](https://github.com/YosysHQ/yosys/files/5755819/test.zip)

tomverbeure, 2020-12-31:
I noticed the same thing. I think it's an artifact of the way signals are being dumped into the VCD file, not a bug in the simulator.

Try the following: only dump waves after the rising edge of the clock. I think you'll see that all values are now correct. (They are for my cases.)

That said: it's probably not a bad idea to understand why exactly this happens.

whitequark, 2020-12-31:
@modwizcode This is a known issue that relates to the way immediately converging designs are treated. You are essentially correct: there *is* a "unexpected half cycle delay". Properly fixing this without losing performance is not trivial though I am working on it; in single-clock designs, what @tomverbeure saying is correct, at the rising edge all of the signals have the right values. As far as I know, if you follow the recommendation in the 

A bit later, I will explain exactly why this happens in CXXRTL's model and what can be done about it.

modwizcode, 2020-12-31:
This makes sense. I look forward to an explanation of the issue. I'm fairly certain I have a slight idea of what's going on here. 

I assume the reason that the vcd doesn't match the printouts is hurried in here somewhere though? (I thought the vcd code and the printouts use the same values so I'm not exactly sure what's going on there either).

I will look at the documentation for the CXXRTL pass again.

I'm hoping to have documented a bit better what the proper way to drive signals on a rising edge are, am I correct in thinking something like set clock high, step, run logic that would run on rising edge, step again, set clock low, step, repeat?

I suspect a hack to fix this might involve running step twice in a row also?

whitequark, 2020-12-31:
You should not drive signals simultaneously (in the same delta cycle) as the rising edge of the clock at all because this introduces a race condition. You can update the inputs, step, update the clock, step again, but this will obviously degrade your performance. This isn't a hack per se but an inherent consequence of the current CXXRTL architecture.

modwizcode, 2020-12-31:
Is there another way to drive signals the same way they'd be driven in say a Verilog testbench with code that runs `@(posedge clk)` then, 50% penalty sounds massive for something that I can't imagine wouldn't be the common case, unless I'm just very confused here. 

In this case I used Verilog translated with CXXRTL to do the real driving to avoid getting this wrong after noticing weird behavior.

I did notice eval and commit are technically split but I'm not sure if that's relevant for this.

whitequark, 2020-12-31:
> Is there another way to drive signals the same way they'd be driven in say a Verilog testbench with code that runs `@(posedge clk)` then

There isn't. You can choose between the performance penalty or the race condition.

> 50% penalty sounds massive for something that I can't imagine wouldn't be the common case

The common case for CXXRTL is simulating a single chunk of synchronous logic such as a CPU and its peripherals. In that case it makes no real difference whether you are driving inputs/sampling outputs at posedge/negedge.

> I might be confused but I know for sure in vhdl when the posedge happens the logic will see a high clock, I don't remember for Verilog exactly but I'd think it crazy to be different.

Like Verilator, CXXRTL implements neither Verilog nor VHDL simulation semantics. This allows it to have performance many orders of magnitude higher than something like iverilog, but in exchange you have to understand and adapt to its simulation semantics.

whitequark, 2020-12-31:
> In this case I used Verilog translated with CXXRTL to do the real driving to avoid getting this wrong after noticing weird behavior.

Depending on how you are doing this (I am not feeling well at the moment so I haven't looked at your code yet), this likely introduces delta cycles, which means your performance penalty is closer to 100%.