YosysHQ/yosys #2525: read_liberty newline handling

thesourcerer8, 2021-01-04:
## Steps to reproduce the issue

Using the file https://github.com/thesourcerer8/caravel_stdcelllib_stdcells/blob/release/cells/lib/XNOR2X1.lib
read_liberty XNOR2X1.lib
1. Executing Liberty frontend.
ERROR: Syntax error in liberty file on line 29.
  Unexpected newline.

When I strip all newlines from the file, it works:
https://github.com/thesourcerer8/caravel_stdcelllib_stdcells/blob/release/cells/lib/libresilicon.lib

From my understanding of the liberty file format specification, it seems that my liberty files are correct, but perhaps I overlooked something?
I would prefer to deliver the liberty files with newlines, to improve readability and debugging.

I have been using Yosys 0.9+2406 (git sha1 f313211c, clang 10.0.0-4ubuntu1 -fPIC -Os) , I am currently trying to reproduce it with current git, when it is finished with compiling.

arcturusannamalai, 2023-05-05:
reproducible on main as of writing

arcturusannamalai, 2023-05-05:
Fix posted; not sure how to add regression

closed 2023-05-21 by mmicko