YosysHQ/yosys #2530: ABC9 flop bypass specify rules can cause combinational loops

Ravenslofty, 2021-01-07:
This testcase has been minimised from a QuickLogic benchmark:
```verilog
module top(input WBs_CLK_i, _1_, _0_, output reg Rx_overflow_detected_o);
  wire rx_fifo_cntr_rst;
  assign rx_fifo_cntr_rst = _1_ | Rx_overflow_detected_o;
  always @(posedge WBs_CLK_i, posedge rx_fifo_cntr_rst)
    if (rx_fifo_cntr_rst) Rx_overflow_detected_o <= 1'h0;
    else Rx_overflow_detected_o <= _0_;
endmodule
```
which is a flop that asynchronously resets itself depending on its output.

## Expected behavior

If you take an ABC flow, such as `synth_ice40`, you get:
```
   Number of wires:                  5
   Number of wire bits:              5
   Number of public wires:           3
   Number of public wire bits:       3
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                  2
     SB_DFFR                         1
     SB_LUT4                         1
```

## Actual behavior

If you take an ABC9 flow, such as `synth_ice40`, you get:
```
2.41.9. Executing ABC9_OPS pass (helper functions for ABC9).
<suppressed ~2 debug messages>
ERROR: Assert `no_loops' failed in passes/techmap/abc9_ops.cc:758.
```

The cause of this is that near-universally flop cells have specify rules to produce an ABC9 flop bypass box that models the combinational path from a set/reset pin to the output. In this particular case, however, this models a combinational loop through the flop, which causes ABC9 to assert.

The reason I'm posting this as an issue is to come to a consensus on how to resolve this given the tradeoffs.

- We could consider this a combinational loop, which is forbidden in RTLIL, keep the specify rules, and possibly extract better results, since a longer critical path means more area reduction opportunities.
- We could remove the specify rules, which I have confirmed fixes the issue, but potentially regress result quality in the rare case that the critical path involves an asynchronous set/reset.
- I could attempt to fix this in ABC9, but with Eddie gone, it's going to be a bug that stays around for a long while, since I am very unfamiliar with the code there.

cc @daveshah1 (for iCE40/ECP5/Nexus), @pepijndevos (for Gowin), @mwkmwkmwk (for Xilinx)

eddiehung, 2021-01-08:
Now now, I wouldn't say I've fallen off the face of the earth. I just don't have the capacity to drive new features forward, but I'm still happy to help.

It looks like you've uncovered a problem with whiteboxes in general, and not specific to flops.
For example, something like:
```
module top(input a, output b);
MUXF7 m(.I0(a), .I1(b), .O(b));
endmodule
```
will trigger it for `synth_xilinx -abc9`.

ABC and ABC9 will break combinatorial loops inside the user design (by arbitrarily cutting the loop into a PI/PO, whether anything better can be done is unclear) before feeding it into `yosys-abc`. However, this SCC analysis (in the `abc9` case, I use `scc` command: https://github.com/YosysHQ/yosys/blob/b0004911ca15ec41db56c76c83461061fedc742e/passes/techmap/abc9.cc#L342) does not currently consider whiteboxes -- I propose that it be made to do so for a more generic solution.

Ravenslofty, 2021-01-08:
That sounds like a much more elegant approach than what I was doing, but as I understand it, whiteboxes by definition have the `whitebox` attribute on them, and my experiments with `scc` suggest there are no cells with that attribute, at least when that line gets called.

daveshah1, 2021-01-08:
Is there a need to deal with greyboxes here too (ie blackboxes with combinational specify rules)?

eddiehung, 2021-01-11:
@daveshah1 Oooh yes. Let me refine my suggestion. I propose that the `scc` command should examine (under a switch) any specify edges defined by blackboxes (and by extension whiteboxes) as it's those specify edges that are ultimately used to define a combinatorial path and delay for ABC to handle. This also makes it easier for the `scc` analysis because it doesn't need to recurse into module.
I admit that my MUXF7 example would otherwise be a pretty stupid thing to do (but not impossible to implement) -- I just wanted to provide an even simpler triggering example -- unlike a flop with a path between its "seq" output and its async input.

eddiehung, 2021-01-11:
@Ravenslofty On Xilinx at least async flops are marked with `(* lib_whitebox *)` which should mean its contents aren't blown away:
https://github.com/YosysHQ/yosys/blob/b0004911ca15ec41db56c76c83461061fedc742e/techlibs/xilinx/cells_sim.v#L716-L717
If I recall, this whitebox attribute is necessary without which `abc9` would not be able to transform this cell into a flop-with-bypass-box where the bypass box contains all the combinatorial (non-edge-sensitive) paths.

Ravenslofty, 2021-01-11:
Okay, I've implemented your proposal, @eddiehung, and my code correctly detects the logic loop through the specify rules on the MUXF7 latch.

However, this still trips `ERROR: Assert `!aig_map.count(bit)' failed in backends/aiger/xaiger.cc:435.`, which indicates there's a combinational loop that ABC9 has not broken.

In the meantime, I'll submit a PR for my SCC patch.

eddiehung, 2021-01-12:
@Ravenslofty I would suggest we make sure that the patch you're proposing is the right approach first. Are you saying that running the MUXF7 example with your changes still triggers that assert?

Ravenslofty, 2021-01-12:
Yes, I am. The patch resolves the `no_loops` assert, but still causes XAIGER to assert.

Ravenslofty, 2021-01-18:
@eddiehung ping

eddiehung, 2021-01-19:
@Ravenslofty Thanks for the reminder!

I've pushed an issue where SCCs were failing to be broken by `write_xaiger` if the output of a box was also an primary output.
Now, rather than making the breaking wire an `inout` (via marking it as `abc9_keep`), I interrupt it using a `$__ABC9_SCC_BREAKER` module instead. I seem to recall that I didn't want to go down this route if I could avoid it, because (a) either it didn't work in the past, or (b) it's assumed to be slightly less efficient.

Amusingly, this modified test-case:
```
module top(output b);
MUXF7 m(.I0(1'b1), .I1(1'b0), .S(b), .O(b));
endmodule
```
which implements a 1-inverter ring oscillator, now gets optimised more directly into an INV by `synth_xilinx -abc9`: presumably because `S -> O` is 296ps, whereas `I -> O` is 127ps.
@Ravenslofty's original test case is now passing too.

Branch here: `eddie/abc9_scc_fixes2` which is based on #2535.

Please confirm it fixes the wider test-case.

Ravenslofty, 2021-01-19:
> Please confirm it fixes the non minimised test-case.

It does not; once again we get an ABC9 `no_loops` assert despite the SCC being flagged. I'll run `bugpoint` once more.

Ravenslofty, 2021-01-19:
```
autoidx 1417
attribute \dynports 1
attribute \hdlname "\\vga_rx_fifo_interface"
attribute \src "vga_rx_fifo_interface.v:30.1-451.10"
module $paramod\vga_rx_fifo_interface\ADDRWIDTH=7\DATAWIDTH=32\VGA_RGB_RXDATA_REG_ADR=5'00110
  parameter \ADDRWIDTH 7
  parameter \DATAWIDTH 32
  parameter \VGA_RGB_RXDATA_REG_ADR 5'00110
  attribute \src "vga_rx_fifo_interface.v:370.1-395.4"
  wire $0\Rx_overflow_detected_pclk[0:0]
  attribute \src "vga_rx_fifo_interface.v:400.1-413.4"
  wire $0\Rx_overflow_detected_r1[0:0]
  attribute \src "vga_rx_fifo_interface.v:400.1-413.4"
  wire $0\Rx_overflow_detected_r2[0:0]
  attribute \src "vga_rx_fifo_interface.v:330.1-367.4"
  wire $0\thresh_line_cnt_reached[0:0]
  attribute $bugpoint 1
  wire input 4 $auto$bugpoint.cc:253:simplify_something$1338
  wire $delete_wire$1065
  wire $delete_wire$1071
  wire $delete_wire$1077
  wire $delete_wire$1083
  wire $delete_wire$1089
  wire $delete_wire$1097
  wire $delete_wire$1099
  wire $delete_wire$1103
  wire $delete_wire$1104
  wire $delete_wire$1105
  wire $delete_wire$1110
  wire width 4 $delete_wire$1120
  wire $delete_wire$1155
  wire $delete_wire$1161
  wire width 11 $delete_wire$1305
  wire $delete_wire$1311
  wire $delete_wire$1312
  wire $delete_wire$1313
  wire $delete_wire$1314
  wire $delete_wire$1323
  wire $delete_wire$1342
  wire $delete_wire$1344
  wire $delete_wire$1346
  attribute \src "vga_rx_fifo_interface.v:254.27-254.80"
  wire $or$vga_rx_fifo_interface.v:254$747_Y
  attribute \src "vga_rx_fifo_interface.v:89.13-89.19"
  wire input 2 \PCLK_i
  attribute \src "vga_rx_fifo_interface.v:130.10-130.35"
  wire \Rx_overflow_detected_pclk
  attribute \src "vga_rx_fifo_interface.v:131.10-131.33"
  wire \Rx_overflow_detected_r1
  attribute \src "vga_rx_fifo_interface.v:132.10-132.33"
  wire \Rx_overflow_detected_r2
  attribute \src "vga_rx_fifo_interface.v:77.26-77.35"
  wire input 1 \WBs_CLK_i
  attribute \src "vga_rx_fifo_interface.v:141.13-141.29"
  wire \rx_fifo_cntr_rst
  attribute \src "vga_rx_fifo_interface.v:129.10-129.33"
  wire \thresh_line_cnt_reached
  attribute \src "vga_rx_fifo_interface.v:93.12-93.37"
  wire output 3 \thresh_line_cnt_reached_o
  attribute \src "vga_rx_fifo_interface.v:254.27-254.80"
  cell $or $or$vga_rx_fifo_interface.v:254$747
    parameter \A_SIGNED 0
    parameter \A_WIDTH 1
    parameter \B_SIGNED 0
    parameter \B_WIDTH 1
    parameter \Y_WIDTH 1
    connect \A $auto$bugpoint.cc:253:simplify_something$1338
    connect \B \Rx_overflow_detected_r2
    connect \Y $or$vga_rx_fifo_interface.v:254$747_Y
  end
  attribute \src "vga_rx_fifo_interface.v:330.1-367.4"
  process $proc$vga_rx_fifo_interface.v:330$764
    attribute \src "vga_rx_fifo_interface.v:333.6-366.8"
    switch \rx_fifo_cntr_rst
      case 1'1
        assign $0\thresh_line_cnt_reached[0:0] 1'0
      case
        attribute \src "vga_rx_fifo_interface.v:349.3-360.6"
        switch $delete_wire$1314
          case 1'1
            assign $0\thresh_line_cnt_reached[0:0] 1'1
        end
    end
    sync posedge \WBs_CLK_i
      update \thresh_line_cnt_reached $0\thresh_line_cnt_reached[0:0]
    sync posedge \rx_fifo_cntr_rst
      update \thresh_line_cnt_reached $0\thresh_line_cnt_reached[0:0]
  end
  attribute \src "vga_rx_fifo_interface.v:370.1-395.4"
  process $proc$vga_rx_fifo_interface.v:370$768
    attribute \src "vga_rx_fifo_interface.v:373.6-394.8"
    switch \rx_fifo_cntr_rst
      case 1'1
        assign $0\Rx_overflow_detected_pclk[0:0] 1'0
      case
        attribute \src "vga_rx_fifo_interface.v:379.3-393.9"
        switch $delete_wire$1312
          case 1'1
            attribute \src "vga_rx_fifo_interface.v:381.7-388.7"
            switch $delete_wire$1104
              case 1'1
                assign $0\Rx_overflow_detected_pclk[0:0] 1'1
            end
        end
    end
    sync posedge \PCLK_i
      update \Rx_overflow_detected_pclk $0\Rx_overflow_detected_pclk[0:0]
    sync posedge \rx_fifo_cntr_rst
      update \Rx_overflow_detected_pclk $0\Rx_overflow_detected_pclk[0:0]
  end
  attribute \src "vga_rx_fifo_interface.v:400.1-413.4"
  process $proc$vga_rx_fifo_interface.v:400$771
    attribute \src "vga_rx_fifo_interface.v:403.3-412.9"
    switch \rx_fifo_cntr_rst
      case 1'1
        assign $0\Rx_overflow_detected_r1[0:0] 1'0
        assign $0\Rx_overflow_detected_r2[0:0] 1'0
      case
        assign $0\Rx_overflow_detected_r1[0:0] \Rx_overflow_detected_pclk
        assign $0\Rx_overflow_detected_r2[0:0] \Rx_overflow_detected_r1
    end
    sync posedge \WBs_CLK_i
      update \Rx_overflow_detected_r1 $0\Rx_overflow_detected_r1[0:0]
      update \Rx_overflow_detected_r2 $0\Rx_overflow_detected_r2[0:0]
    sync posedge \rx_fifo_cntr_rst
      update \Rx_overflow_detected_r1 $0\Rx_overflow_detected_r1[0:0]
      update \Rx_overflow_detected_r2 $0\Rx_overflow_detected_r2[0:0]
  end
  connect \rx_fifo_cntr_rst $or$vga_rx_fifo_interface.v:254$747_Y
  connect \thresh_line_cnt_reached_o \thresh_line_cnt_reached
end
```

This is what bugpoint has spat out this time, and it's very odd. It crashes the `synth_quicklogic -abc9` flow I've been working on, as well as `synth_ice40 -abc9` and `synth_ecp5 -abc9`, but *not* `synth_xilinx -abc9` or `synth_intel_alm` (`synth_gowin -abc9` crashes for unrelated reasons).

Ravenslofty, 2021-01-20:
```verilog
module top (output reg _0_);
  reg Rx_overflow_detected_r2;
  wire rx_fifo_cntr_rst;
  always @(posedge 1'hx, posedge rx_fifo_cntr_rst)
    if (rx_fifo_cntr_rst) _0_ <= 1'h0;
    else _0_ <= 1'hx;
  always @(posedge 1'hx, posedge rx_fifo_cntr_rst)
    if (rx_fifo_cntr_rst) Rx_overflow_detected_r2 <= 1'h0;
    else Rx_overflow_detected_r2 <= 1'hx;
  assign rx_fifo_cntr_rst = 1'hx | Rx_overflow_detected_r2;
endmodule
```
Here's my present minimal example, using the following bare ECP5 synthesis script:
```
        read_verilog -lib -specify +/ecp5/cells_sim.v
        proc
        techmap
        dfflegalize -cell $_DFF_?_ 01 -cell $_DFF_?P?_ r
        techmap -D NO_LUT -map +/ecp5/cells_map.v
        abc9
```

eddiehung, 2021-01-25:
Latest test-case seems top show that no SCCs were detected when I would expect one:
```
7.9. Executing SCC pass (detecting logic loops).
Found 0 SCCs in module top.
Found 0 SCCs.

7.10. Executing ABC9_OPS pass (helper functions for ABC9).
<suppressed ~2 debug messages>
ERROR: Assert `no_loops' failed in passes/techmap/abc9_ops.cc:768.
```
thus `abc9` had nothing to break.

Inserting a `show` after `abc9`'s `scc` gives:
[image]

Ravenslofty, 2021-01-25:
It turns out that I forgot to `sigmap` my ports in the `scc-specify` branch, so the output of the OR gate and `rx_fifo_cntr_rst` weren't being considered the same wire. That's now been fixed, and you'll probably want to rebase your branch. The minimal Verilog test I gave above now works, but the RTLIL testcase still crashes with `no_loops` despite `scc` now detecting the loop.

Ravenslofty, 2021-01-25:
```verilog
module \$paramod\vga_rx_fifo_interface\ADDRWIDTH=7\DATAWIDTH=32\VGA_RGB_RXDATA_REG_ADR=5'00110 (thresh_line_cnt_reached_o);
  reg Rx_overflow_detected_r2;
  wire rx_fifo_cntr_rst;
  output thresh_line_cnt_reached_o;
  reg thresh_line_cnt_reached_o;
  assign rx_fifo_cntr_rst = 1'hx | Rx_overflow_detected_r2;
  always @(posedge 1'hx, posedge rx_fifo_cntr_rst)
    if (rx_fifo_cntr_rst) Rx_overflow_detected_r2 <= 1'h0;
    else Rx_overflow_detected_r2 <= 1'hx;
  always @(posedge 1'hx, posedge rx_fifo_cntr_rst)
    if (rx_fifo_cntr_rst) thresh_line_cnt_reached_o <= 1'h0;
    else thresh_line_cnt_reached_o <= 1'hx;
endmodule
```

Minimised to use the following script:
```
        read_verilog -lib -specify +/ecp5/cells_sim.v +/ecp5/cells_bb.v
        proc
        techmap -map +/techmap.v
        dfflegalize -cell $_DFF_?_ 01 -cell $_DFF_?P?_ r
        techmap -D NO_LUT -map +/ecp5/cells_map.v
        abc9
```

There seem to be a lot of similar but subtle bugs lurking around this testcase...

Ravenslofty, 2021-02-01:
@eddiehung ping

arblake, 2021-02-24:
Is there a fix ??? I too get :
2.44.15.1. Executing ABC9_OPS pass (helper functions for ABC9).

2.44.15.2. Executing ABC9_OPS pass (helper functions for ABC9).

2.44.15.3. Executing XAIGER backend.
ERROR: Assert `!aig_map.count(bit)' failed in backends/aiger/xaiger.cc:435.

eddiehung, 2021-03-16:
Sorry for the delay. Took a quick look and it looks like there are two SCCs:

[image]

However `scc -specify` only identifies the red one (I don't know if this is the intended behaviour) which means only the red loop is broken and blue remains:

[image]

eddiehung, 2021-03-16:
Okay, after trying a simple example I now think `scc` won't pick up both loops. My approach now is to block bypass-ing for cells that are on a comb loop. This fixes all three testcases here. Please try the `eddie/abc9_scc_fixes2` branch again.

PR #2694 (2021-03-30): abc9: fix SCC issues
Fix #2530 -- specifically `(* abc9_box, lib_whitebox *)` cells that can be bypassed using a combinatorial-only box are no longer bypassed if they form part of a combinatorial loop (based on analysing connectivity only, not if it such a condition can be reached or will cause undefined behaviour -- such are the testcases from #2530 which occurs when the output of an async flop (which causes a bypass cell to capture async behaviour) fans back into to its async set/reset input).

Improve SCC handling overall. This is achieved by no longer marking combinatorial loops with `(* a [... cut]
closed 2021-03-30 by eddiehung