YosysHQ/yosys #2531: Yosys errors out when reading liberty file

sjalloq, 2021-01-07:
## Steps to reproduce the issue

I'm working with a proprietary lib file so can't send the full contents.  But hopefully I can provide enough info to help identify the issue.

## Expected behavior

Read the liberty file with no errors.

## Actual behavior
```
yosys> read_liberty some.lib
1. Executing Liberty frontend.
ERROR: Missing function on output ENL of cell TLATNCA_X12_A7TR_W3.
```

The pin definition of ENL is:

```  
pin (ENL) {
   direction : "internal";
   internal_node : "ENL";
   related_power_pin : "VDD";
   related_ground_pin : "VSS";
   related_bias_pin : "VNW VPW";
  }
```
The cell in question is a clock gating latch with inputs CK, E and an output ECK. The ENL pin is defined by a statetable and is used in the state_function of the output pin ECK, `state_function : "CK * ENL";`.

Not being versed in liberty format, I can only assume that the tool is mistaking an "internal" node for an "output".  Thoughts?

Ravenslofty, 2021-01-11:
The "function" here is actually the direction attribute. Yosys doesn't care about internal pins; and the easiest solution is to just remove them.

ted-xie, 2021-05-04:
> the easiest solution is to just remove them

If yosys indeed doesn't care about internal pins, shouldn't those pins be ignored, rather than causing a synthesis failure?

A colleague and I came across this behavior today while trying to synthesize with a library whose ICG cells all contain internal pins.

closed 2021-08-13 by mmicko