YosysHQ/yosys #2540: assert in write_cxxrtl

galibert, 2021-01-14:
m68k.v is the result of sv2v on fx68k

Do:
read -vlog95 m68k.v
hierarchy -top fx68k
write_cxxrtl m68k.cc

The write_cxxrtl step asserts in dump_sigchunk with a WireType::UNUSED that is not lhs.
ERROR: Assert `is_lhs' failed in backends/cxxrtl/cxxrtl_backend.cc:897.

[m68k.zip](https://github.com/YosysHQ/yosys/files/5815511/m68k.zip)

galibert, 2021-01-14:
[mem.zip](https://github.com/YosysHQ/yosys/files/5815589/mem.zip)
Missing mem files

modwizcode, 2021-01-14:
I've looked at this a bit and the issue appears to be that during `write_cxxrtl` it detects the wire as unused at optimization level `-O5` or above. Those allow "public" wires (iirc this is Yosys slang for "named in the design but not an output), to be marked as `LOCAL` which means they're not exposed outside the `eval()` method for CXXRTL.

This later leads to the wire in question (in this case `nDecoder.pcRelDbl`) being marked as `UNUSED` because it ends up being able to be totally optimized out of the design (not 100% sure about the details on how that optimization takes place, but I'm assuming it's valid and that the tree of wires it connects to eventually terminates as unused). 

Because the wire is UNUSED, this means that nothing should assign to it ever leading to the reason for the failing assert to exist. However, during `-g4` (the highest debug info setting), debug info is created for *all* public signals in the design, including those that are optimized out. In order to do this it actually generates a secondary `eval()` method that includes such unoptimized signals in the design, running through the passes but with `for_debug = true`. 

This shouldn't fail because theoretically there's a separate `WireType` mapping for the wires in debug mode, where `UNUSED` should be mapped to something else.  I think the actual issue is the use of `||` instead of `&&` here
https://github.com/YosysHQ/yosys/blob/01626e6746e85832d5b75785b9e45f39ff8eb299/backends/cxxrtl/cxxrtl_backend.cc#L2650

Since some wires will be in one set but not the other, and that check actually tests for the wire being unused in both sets. Patching that seems to resolve the issue and I'll submit a PR.

PR #2871 (2021-07-16): cxxrtl: don't mark buffered internal wires as UNUSED for debug
Public wires may alias buffered internal wires, so keep `BUFFERED` wires in debug information even if they are private. Debug items are only created for public wires, so this does not otherwise affect how debug information is emitted.

Fixes #2540.
Fixes #2841.
Supersedes #2541.
commit 44a3d924 (2021-07-16): cxxrtl: don't mark buffered internal wires as UNUSED for debug.

Public wires may alias buffered internal wires, so keep BUFFERED
wires in debug information even if they are private. Debug items are
only created for public wires, so this does not otherwise affect how
debug information is emitted.

Fixes #2540.
Fixes #2841.
closed 2021-07-16 by whitequark