YosysHQ/yosys #2542: cxxrtl seems to lose some signals somehow

galibert, 2021-01-14:
With the same verilog as Issue #2540 the generated C++ seems incorrect, no matter the -O level in write_cxxrtl.

Steps to reproduce
  read -vlog95 m68k.v
  hierarchy -top fx68k
  write_cxxrtl m68k.cc

The design (and the 68000 itself) includes a free-running lfsr-based divided-by-10 clock that is output on pin E.  The lfsr is called eCntr.  Looking at the C++ shows that nothing ever writes to p_E or p_eCntr which is not good at all.
[m68k.zip](https://github.com/YosysHQ/yosys/files/5816582/m68k.zip)

galibert, 2021-01-14:
Note that requires the #2540 fix first, of course.

modwizcode, 2021-01-14:
This issue is caused by cxxrtl somehow not paying attention to the specific clock bit in the vector being used when flops (and possibly other things) are clocked such as `@(posedge Clks[4])`, while this is arguably kind of insane in my opinion, it might make sense coming out of a code generator.

A workaround is that the value can be first assigned to a wire and then used to clock the process which will result in yosys assigning the wire to the flops, which will be be correctly handled in cxxrtl. Due to optimizations in yosys it's not a "safe" workaround as a pass could easily see through the indirection and express the connection directly. In this case it will work though. (cxxrtl for what it's worth, will also see through the indirection and extract the internal clock, even on -O0, it just doesn't understand directly clocking off of the signal when it runs it's initial passes).

I'll try to track down where that info is being ignored so this can be fixed.

modwizcode, 2021-01-14:
This issue seems to be caused by this line: https://github.com/YosysHQ/yosys/blob/01626e6746e85832d5b75785b9e45f39ff8eb299/backends/cxxrtl/cxxrtl_backend.cc#L1113

The reason this happens is that `SigSpec::is_wire()` is defined as: https://github.com/YosysHQ/yosys/blob/ec410c9b1934c5c06ad6f71e60f2337fabe5b055/kernel/rtlil.cc#L4060

which requires the underlying wire to have a width the same as the `SigSpec` itself. In this case that's not true, the `SigSpec` for the attached clock input is only a single bit slice of the full signal width. I'm not sure why `is_wire()` should only be true if it's not a partial slice. I don't know enough about the internals of Yosys design to understand if that's intentional.

What this means is that when the clocks are scraped from a vector, we only want the first two checks, since the input clock might only be a bit portion. This should be fine since clocks will only ever be a single bit, and we convert to a `SigBit` anyway before trying to map that bit to an underlying signal. 

I'm going to submit a PR which fixes this by only checking if that signal is `is_bit()` which only tests for the width of the `SigSpec` being 1, since `@(posedge)` already requires a single bit, this requirement should always be met.

modwizcode, 2021-01-14:
After looking at some other lines I think that this issue with clock inputs affects cxxrtl more broadly since most every clock is tested with `is_wire()` which will fail if it's from a bit vector. I don't know that `is_wire()` can be redefined either since the fact that `is_wire()` means a `SigSpec` is 1:1 matched to it's `Wire` backing might be a dependency of other code, causing passes to potentially break in subtle ways.

Instead I see two options:
1. Define an `is_slice()` or something on `SigSpec` that tests for everything except width being the same
2. Make a local cxxrtl internal method to do that test

I'm going with 2 for now because I don't want to potentially hold this fix up with deciding if that change is a good one to make.

PR #2544 (2021-01-26): CXXRTL: Fix sliced bits as clock inputs
Currently in cxxrtl due to an earlier change that fixed issues with constant signals driving clock inputs of clocked cells, when cells are clocked by a `SigSpec` that doesn't pass `is_wire()` most of them will be ignored. 

This causes an issue because slices of another signal (i.e. a single `SigBit`) don't return `true` for `is_wire()`. While it's arguable that there should be some `is_wire()` like function defined that passes for slices (at least for single bits), that's another issue.

In order to fix #2542, I implemented a subset of `is_wire()` inter [... cut]
commit 86607d0f (2021-01-26): Handle sliced bits as clock inputs (fixes #2542)
closed 2021-01-26 by whitequark