YosysHQ/yosys #2545: Allow loading plugins "by name" from more than one source (alt: add a search path for plugins)

thoughtpolice, 2021-01-17:
When yosys is invoked with the flag `-m foo` or the command `plugin -i foo` is run &mdash; that is, with a non-absolute path to a dynamically loadable library &mdash; the loader will look for a file named `foo.so` by default in (among other places) `/usr/share/yosys/plugins`, where `$PREFIX/share/yosys` is also known as `YOSYS_DATDIR`.

The intention of this design is that this single location is global and shared across the system; tools like **[yosys-ghdl](https://github.com/ghdl/ghdl-yosys-plugin)** or **[yosys-bluespec](https://github.com/thoughtpolice/yosys-bluespec)** should by default be installed into such a location by invoking `cp ... $(yosys-config --datdir)/plugins` or whatever after compilation. Package managers should put the `.so` files in this directory, and so forth.

This is mostly true, except when using a package manager like **[Nix](https://nixos.org)**. To be very brief, Nix installs a package into a directory, and then marks that directory as read-only. This means that plugins cannot, later on, install their `.so` files into the same data directory of the yosys package. Nix users, therefore, can't simply install a plugin and then invoke a command like `yosys -m bluespec -p 'read_bluespec design.bsv'` &mdash; the location of the plugin file will exist in another directory.

---

To work around this as the Yosys maintainer for Nix, I've added a patch to Yosys that effectively makes yosys instead search a *list* of directories when looking for a "named" plugin. This patch **[is very simple](https://github.com/NixOS/nixpkgs/blob/2660037f8cdf6bb69bacbf3d9d2f9f72d056c8d6/pkgs/development/compilers/yosys/plugin-search-dirs.patch)**. It allows you to add a set of directories to the environment variable `NIX_YOSYS_PLUGIN_DIRS` separated by `:`, and Yosys will search them. So invoking

```bash
$ export NIX_YOSYS_PLUGIN_DIRS=/d1:/d2:...
$ yosys -m bluespec
```

will try to load `bluespec.so` from the following locations, in this order:

```
YOSYS_DATDIR/plugins/bluespec.so
/d1/bluespec.so
/d2/bluespec.so
...
```

(Note this environment variable is private and undocumented. It is populated automatically by the Nix infrastructure, so users should be unaware of it. It only exists as a means to an end to make the Yosys UX for Nix users behave similarly to non-Nix users.)

---

The specifics of this issue aren't really about Nix, though &mdash; the above situation could apply to a number of other scenarios i.e. a non-root user using a root-installed yosys, who wants to compile and load their own plugins. So in general, it would be nice to have a more general, proper solution implemented upstream, if it's welcome.

whitequark, 2021-01-17:
Sounds perfectly reasonable to me to have something like `YOSYS_PLUGIN_DIRS`. The main point I'd like to raise is that it should work on Windows and respect its conventions there.

Xiretza, 2021-01-22:
This would also be useful for running the testsuite for https://github.com/SymbiFlow/yosys-symbiflow-plugins during packaging. For the implementation I would however suggest searching the directories from the environment variable BEFORE `YOSYS_DATDIR`, such that it's possible to override already-installed plugins.

widlarizer, 2026-11-04:
Fixed in #5427

closed 2026-11-04 by widlarizer