YosysHQ/yosys #2555: Function call in function call argument not supported

gs-jgj, 2021-01-23:
## Steps to reproduce the issue

Compile the following code:
```verilog
module test_functions(input clk,
                      input [31:0] a,
                      input [31:0] b,
                      output reg [31:0] sum);

    function automatic [15:0] re(input [31:0] x);
        re = x[15:0];
    endfunction

    function automatic [15:0] im(input [31:0] x);
        im = x[31:16];
    endfunction

    function automatic [31:0] complex(input [15:0] r, input [15:0] i);
        complex = {i, r};
    endfunction

    function automatic [15:0] fp16_add_saturate(input [15:0] a, input [15:0] b);
        reg [16:0] result;
        begin
            result = {a[15],a} + {b[15],b};
            if (result[16:15] == 2'b10)
                fp16_add_saturate = {1'b1, {15{1'b0}}};
            else if (result[16:15] == 2'b01)
                fp16_add_saturate = {1'b0, {15{1'b1}}};
            else
                fp16_add_saturate = result[15:0];
        end
    endfunction

    function automatic [31:0] add_complex(input [31:0] a, input [31:0] b);
        add_complex = complex(fp16_add_saturate(re(a), re(b)),
                              fp16_add_saturate(im(a), im(b)));
    endfunction

    always @(posedge clk)
        sum <= add_complex(a, b);

endmodule
```

## Expected behavior

It compiles and works :) 

## Actual behavior

Returns the error `Non-constant function call in constant expression.` in the add_complex function body

It's not entirely clear what triggers this problem since some levels of nested function calls seem to work sometimes

zachjs, 2021-01-28:
I believe this is the same issue discussed in #2562. I have a pair of changes with a fix posted there pending review.

whitequark, 2021-02-03:
Closing as duplicate of #2562.

closed 2021-02-03 by whitequark