YosysHQ/yosys #2561: How to generate area, power and timing report

rahulraveendran15-coder, 2021-01-26:
Hi,
How to generate area,power and timing report for the design synthesized through Yosys. I am new to Yosys.

Ravenslofty, 2021-01-26:
Yosys can't actually perform any of this analysis by itself (...mostly; see #2490 for an FPGA-oriented static timing analyser). The level that Yosys works at is a bit too abstract to handle these things; interconnect delays can very much change the final result, but Yosys doesn't know of this.

rahulraveendran15-coder, 2021-01-27:
@Ravenslofty 
But in Ibex synthesis script I can see below code which generated the area report. But in Yosys manual it doesn't mention anything like this. So how Ibex able to generate area report using Yosys

`yosys "tee -o $lr_synth_out_dir/reports/area.rpt stat -liberty $lr_synth_cell_library_path"`

Reference [:](https://github.com/lowRISC/ibex/blob/master/syn/tcl/yosys_run_synth.tcl)

closed 2021-08-13 by mmicko