YosysHQ/yosys #2562: Apparent bug in const checking logic

rswarbrick, 2021-01-26:
To reproduce this issue, check out [Ibex](https://github.com/lowRISC/ibex), e.g. version b295f07b (although this shouldn't be version-sensitive) and then run the synthesis flow. The instructions in `syn/README.md` should work.

The synthesis flow works by running sv2v and then running Yosys on the output. With the current head of the Yosys master branch, you get the following error message:
```
out/generated/ibex_cs_registers.v:544: ERROR: Non-constant function call in constant expression.
```
The relevant code looks like this:
```systemverilog
mstatus_d = {
   ... // some other elements in the concatenation
   sv2v_cast_2(sv2v_cast_2(csr_wdata_int[ibex_pkg_CSR_MSTATUS_MPP_BIT_HIGH:ibex_pkg_CSR_MSTATUS_MPP_BIT_LOW])),
   ... // some other elements in the concatenation
};
```
Here, `sv2v_cast_2` is defined as follows:
```systemverilog
	function automatic [1:0] sv2v_cast_2;
		input reg [1:0] inp;
		sv2v_cast_2 = inp;
	endfunction
```

Bisecting Yosys builds points to commit 2085d9a.

If you manually hack the generated code to replace the argument to the inner `sv2v_cast_2` with a constant like `8'b0` OR (more confusingly) remove one layer of `sv2v_cast_2` calls, the error message goes away.

@zachjs (since this was your change, and you probably also understand the output from sv2v better than me!)

zachjs, 2021-01-26:
Short version: The bug has to do with faulty logic checking for recursive functions. I have a fix locally, but the fix depends on my earlier PR #2529.

`has_const_only_constructs` attempts to check for recursive calls to encourage constant evaluation. There is logic from an earlier change of mine that attempts constant evaluation of any arguments to a function call where constant evaluation is recommended. I originally corrected the recursive function detection, but I noticed further issues while adding test coverage.

My proposal is to _attempt_ to use constant function evaluation on any function with all-const arguments. In cases where constant evaluation is truly required, it can fail as before. In other cases, it can fall back on the non-const function elaboration. This requires some changes to short circuit the failure cases of `eval_const_function`, but otherwise this simplifies the frontend, in my opinion.

rahulraveendran15-coder, 2021-01-27:
@zachjs 
Could you please update once you have the proper fix for this, so that I can build Yosys again with new release and try running synthesis for Ibex

zachjs, 2021-01-27:
@rahulraveendran15-coder My branch https://github.com/zachjs/yosys/tree/repeat-call contains both the (recently updated) changes in #2529 as well as my above suggested fix.

rahulraveendran15-coder, 2021-01-28:
@zachjs 
Its not working the same error is getting displayed, I tried below commands could you please tell me if I am wrong.
```
git clone https://github.com/zachjs/yosys.git
cd yosys
mkdir build
cd build
make -f ../Makefile
```

rahulraveendran15-coder, 2021-01-28:
@zachjs 
Thanks, it is working with the branch.
 When this fix will be merged to yosys main git repo, so that it will be available to everyone

zachjs, 2021-01-28:
@rahulraveendran15-coder I'm not sure when these changes will be merged. It's possible that concerns will be raised once it is reviewed. While my preference would be to get these fixes merged, we could consider temporarily rolling back part of the recursive functions change in the near term.

PR #2573 (2021-02-11): verilog: refactored constant function evaluation
Elaboration now attempts constant evaluation of any function call with only constant arguments, regardless of the context or contents of the function. This removes the concept of "recommended constant evaluation" which previously applied to functions with `for` loops or which were (sometimes erroneously) identified as recursive. Any function call in a constant context (e.g., `localparam`) or which contains a constant-only procedural construct (`while` or `repeat`) in its body will fail as before if constant evaluation does not succeed.

Fixes #2 [... cut]
closed 2021-02-11 by whitequark