YosysHQ/yosys #2570: 2D Array assignment in port does not work

joshtyler, 2021-01-31:
2D array assignment doesn't seem to be working when assigned in a module port.

## Steps to reproduce the issue

1. Download and extract [broken.zip](https://github.com/YosysHQ/yosys/files/5900089/broken.zip). It contains one file broken.sv.
2. Run `yosys -p "read_verilog -sv broken.sv; synth_ice40; write_verilog"`
3. Observe the value assigned to `out`

I have rebuilt from master, and verified that this behaviour still occurs, my version is:
`Yosys 0.9+3882 (git sha1 beeaad19, clang 11.0.0 -fPIC -Os)`

Here are the contents of broken.sv for reference:
```
module intermediate
(    
    output logic [3:0] out
);
    assign out = 4'hA;
endmodule

module broken
(
    output logic [3:0] out
); 
    // Works - out is assigned 4'hA
    //logic [3:0] vec [0:0];
    //assign vec[0] = 4'hA;
    //assign out = vec[0];

    // Works - out is assigned 4'hA
    //logic [3:0] vec;
    //intermediate inst (.out(vec[3:0]));
    //assign out = vec[3:0];

    // Doesn't work - out is assigned 4'hX
    logic [3:0] vec [0:0];
    intermediate inst (.out(vec[0]));
    assign out = vec[0];
endmodule
```

## Expected behavior
`out` is assigned is assigned 4'hA.

## Actual behavior

`out` is assigned is assigned 4'hX.

zachjs, 2021-02-04:
TLDR: The frontend elaborates `.out(vec[0])` into an `AST_MEMRD` as it has really has no logic for treating expressions bound to outputs as LHSs, and in turn generating an `AST_MEMWR` instead. Possible workarounds include A) forcing `-mem2reg`; or B) indirecting the assignment through a temporary, i.e.,
```SystemVerilog
logic [3:0] vec [0:0];
logic [3:0] tmp;
intermediate inst (.out(tmp));
always @* vec[0] = tmp;
assign out = vec[0];
````

The port direction information isn't technically available at the point where the memory access is being turned into an `AST_MEMRD`. The `mem2reg` logic would ideally also be able detect such a binding as an assignment to a memory. At a high level, either the direction information of the underlying module needs to be made available, or downstream phases could handle resolving the ambiguity once that information becomes available.

This problem is reminiscent of the elaboration ordering considerations involved in supporting interfaces, as well as those raised by the work on supporting hierarchical connections. While port directionality information would be relatively easy to gather without elaborating the underlying module, I believe other issues and requests may also run intro trouble with the current "elaborate-then-instantiate" strategy. I imagine other tools work by evaluating constant expressions within a module, instantiating cells it contains, and then proceeding with the full elaboration. I would also guess that this is why the LRM does not allow the use of hierarchical identifiers within constant expressions. I'm curious what the more seasoned contributors think about all this.

whitequark, 2021-02-04:
Thanks for the analysis @zachjs! I don't have a deep enough understanding of the frontend to have an opinion myself. I suspect only @clairexen could.

zachjs, 2021-02-06:
I just stumbled across https://github.com/YosysHQ/yosys/issues/1053#issuecomment-619420890, which raises similar concerns regarding the current elaboration strategy.