YosysHQ/yosys #2571: Yosys issue error in propagating parameter from one module to other module: ERROR: implicitly declared outside of a module.

dineshannayya, 2021-02-03:
############## File: a1.sv  ##########################

include "a.svh"
module a1 (
    input   logic                                   rst_n,                      
    input   logic                                   clk                        
);
endmodule

############## File: b1.sv  ##########################

`include "a.svh"
`include "b.svh"

module b1 (
    input   logic         rst_n,      
    input   logic         clk        

);
endmodule

############  File:  a.svh ###################

`ifndef INCLUDE_A_DEFS
`define INCLUDE_A_DEFS

  parameter bit [31:0] TRIG_NUM = 4; 

`endif 

########## File: b.svh ##########################

`ifndef INCLUDE_B_DEFS
`define INCLUDE_B_DEFS

   parameter bit [31:0] MTRIG_NUM             = TRIG_NUM; 

`endif 

############  yosus Run Log ########################
 Yosys 0.9+3871 (git sha1 98afe2b7, clang 10.0.0-4ubuntu1 -fPIC -Os)

yosys> read_verilog -sv a1.sv 
1. Executing Verilog-2005 frontend: a1.sv
Parsing SystemVerilog input from `a1.sv' to AST representation.
Generating RTLIL representation for module `\a1'.
Successfully finished Verilog frontend.

yosys> read_verilog -sv a1.sv 

2. Executing Verilog-2005 frontend: b1.sv
b.svh:4: ERROR: Identifier `\TRIG_NUM' is implicitly declared outside of a module.

#########################################

## Steps to reproduce the issue
Run the yosys in following step

read_verilog -sv a1.sv
read_verilog -sv b1.sv 

b.svh:4: ERROR: Identifier `\TRIG_NUM' is implicitly declared outside of a module.

## Expected behavior
Expected behaviour was global parameter defined in previous file should have propagated to next file

## Actual behavior
Tool is issue warning:
ERROR: Identifier `\TRIG_NUM' is implicitly declared outside of a [module.]

[issue.tar.gz](https://github.com/YosysHQ/yosys/files/5916954/issue.tar.gz)

zachjs, 2021-02-06:
I think this may be the same issue described in #2440, where the preprocessor definitions are preserved across files, but the global definitions are not.

whitequark, 2021-02-06:
Right. Aren't enums and parameters tracked separately though?

zachjs, 2021-02-06:
They are tracked separately, but I'd argue that changes to the compilation unit behavior for any of {preprocessor definitions, global declarations, typedefs, default_nettype, etc.} would ideally apply to all of them to avoid too much confusion.

zachjs, 2021-03-12:
In looking through this issue a second time, the actual problem is that globals in one file cannot depend on globals in another file. I believe bug this was accidentally introduced as part of #1642. I have a fix incoming.

closed 2021-03-14 by whitequark

dineshannayya, 2021-03-15:
Thanks, In the latest yosys release i see this issue is fixed.  But I still see issue in Propagating Gobal Typedef from one file to other file. I will be raising this in separate issue.

zachjs, 2021-03-15:
@dineshannayya I believe an issue for that already exists: https://github.com/YosysHQ/yosys/issues/2440

dineshannayya, 2021-03-15:
Yes this looks to be same issue

PR #2658 (2021-03-14): sv: allow globals in one file to depend on globals in another
This defers the simplification of globals so that globals in one file may depend on globals in other files. Adds a simplify() call downstream because globals are appended at the end.

Fixes #2571