YosysHQ/yosys #2575: sig.as_wire()->is_signed' failed

mehdimolum, 2021-02-05:
Hi I am getting following error and I don't have any idea which part of my code causing this error. 
Could you please advise me how to fix this?

> Warning: Replacing memory \cnt_CHKN_proc with list of registers. See RTL/Decode.sv:410
> ERROR: Assert `arg->is_signed == sig.as_wire()->is_signed' failed in frontends/ast/genrtlil.cc:1738.

zachjs, 2021-02-05:
Could you share the input which yields this error, as well as your Yosys version (`yosys -V`)?

mehdimolum, 2021-02-05:
the version/system info:
Yosys 0.9+3885 (git sha1 baf18753, gcc 9.3.0-17ubuntu1~20.04 -fPIC -Os)

and here is the failing .ys

> read_verilog -sv RTL/Decode.sv ../Init/RTL/LDPC_Dec_Init.sv ../CHKNODE/CHKN_Single/RTL/CHKN_Single.sv ../CHKNODE/CHKN_Bank/RTL/CHKN_Bank.sv ../Add_Saturate/RTL/Add_Saturate.sv
> hierarchy -check -top Decode
> proc; opt; fsm; opt; #memory; opt 
> 

unfortunately I can't share the code but it is some lengthy code so I am not even sure which part of code causing this. 

In the meantime, I used to synthesize it with previous installation of yosys. I am going to install the same version and try again

zachjs, 2021-02-05:
After some additional poking around, I found a bug (fix is in #2576), but it may not be the bug you are encountering. If that change does not fix the issue for you, we can work through some steps to identify the issue without sharing any private source.

mehdimolum, 2021-02-05:
unfortunately it didn't fix the error, except it is in line 1739 now

zachjs, 2021-02-05:
I'm interested in seeing the AST dump for the problematic binding. The AST dump may also refer to a source location for the expression. To produce this, we could add just before the failing assertion:

```C++
if (arg->is_signed != sig.as_wire()->is_signed) {
    arg->dumpAst(nullptr, "> ");
}
```

mehdimolum, 2021-02-05:
the log output is 

> AST_MEMRD <RTL/Decode.sv:180.12-180.18> [0x2df51f0 -> 0x2e21d50] str='\AS_o_c' signed basic_prep
> >   AST_CONSTANT <RTL/Decode.sv:180.19-180.32> [0x42d7610] bits='00000000000000000000000000000000'(32) signed basic_prep range=[31:0]
> ERROR: Assert `arg->is_signed == sig.as_wire()->is_signed' failed in frontends/ast/genrtlil.cc:1865.

and if I am not wrong, I think it is this related to `AS_o_c` that is connected to port `o_c`. following is port definition

> module Add_Saturate #(parameter N = 3)
> 		(
> 		input wire                i_reset,
> 		input  wire signed [N-1:0]i_a,
> 		input  wire signed [N-1:0]i_b,
> 		output reg signed [N-1:0]o_c
> 		);

that is instantiated as follows:

> logic signed [WIDTH_L-1:0]                        AS_i_a [0:Z-1][0:N_L_IN_OUT-1];                                               
> 	logic signed [WIDTH_L-1:0]                        AS_i_b [0:LEN_CW-1];
> 	logic signed [WIDTH_L-1:0]                        AS_o_c [0:LEN_CW-1];
> 	generate 
> 		for (i_Col = 0; i_Col < 16; i_Col++)begin //TODO : right value instead of 16 (probably N_CAL_MAX)
> 			for (i_Z = 0; i_Z < Z; i_Z++)begin
> 				Add_Saturate #(.N(WIDTH_L)) UUT2_Add_Saturate(
> 						.i_reset(i_reset),
> 						.i_a(AS_i_a[i_Z][i_Col]),
> 						.i_b(AS_i_b[i_Col*Z + i_Z]),
> 						.o_c(AS_o_c[i_Col*Z + i_Z]));
> 			end
> 		end
> 	endgenerate

zachjs, 2021-02-06:
@mehdimolum Thank you for providing that information! I've updated #2576 to fix the issue with signed memory reads and a couple of other closely related issues. Please let me know if it fixes your issue!

mehdimolum, 2021-02-06:
great! thats fixed now

PR #2576 (2021-02-06): genrtlil: fix signed port connection codegen failures
This fixes binding signed memory reads, signed unary expressions, and signed complex SigSpecs to ports. This also sets `is_signed` for wires generated from signed params when -pwires is used. Though not necessary for any of the current usages, `is_signed` is now appropriately set when the `extendWidth` helper is used.

Fixes #2575
closed 2021-02-06 by whitequark