YosysHQ/yosys #2579: verilog: regression: localparam not recognized as constant

Xiretza, 2021-02-09:
## Steps to reproduce the issue

```verilog
module test;
parameter X = 1'b0;

if (X == 1'b1) begin
	localparam Y = 1'b1;
end else begin
	localparam Y = 1'b0;
end

child #(.Y(Y)) _ ();
endmodule
```

This is a simplified reproducer of [a symbiflow techmap](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/45c83a2592923b9a785aa05c9464b5f453ebd151/xc/xc7/techmap/cells_map.v#L862) that started failing recently.

```
$ yosys -p 'read_verilog /tmp/test.v'
[...]
 Yosys 0.9+3894 (git sha1 eff18a2b1, gcc 10.2.0 -march=x86-64 -mtune=generic -O2 -fno-plt -fvar-tracking-assignments -fdebug-prefix-map=/build/yosys-git/src=/usr/src/debug -fPIC -Os)

-- Running command `read_verilog /tmp/test.v' --

1. Executing Verilog-2005 frontend: /tmp/test.v
Parsing Verilog input from `/tmp/test.v' to AST representation.
Generating RTLIL representation for module `\test'.
/tmp/test.v:10: ERROR: Parameter _.Y with non-constant value!
```

This works correctly in 708eb327a, I'm assuming it's related to #2529.

## Expected behavior

The verilog file can be read successfully.

## Actual behavior

The verilog frontend errors.

zachjs, 2021-02-09:
This given example is unfortunately illegal. `Y` is declared within a generate block (implicitly named `genblk1`) and is not visible at the module's top-level scope. Every simulator I can get my hands on correctly rejects this example. #2529 removed the previous invalid scoping behavior in Yosys. While it is possible to refer to `Y` as `genblk1.Y`, the LRM forbids the use of hierarchical identifiers as constant primaries. Note that Yosys (on master) and at least one suite of commercial tools allow the use of hierarchical identifiers as constant primaries in certain scenarios (including this one), but I wouldn't recommend depending on this non-standard support.

whitequark, 2021-02-09:
Closing as intended behavior. If this turns out to have too much fallout, we can revisit this and, perhaps, add a workaround of some kind.

closed 2021-02-09 by whitequark

Xiretza, 2021-02-09:
No problem, thanks for the quick response!