YosysHQ/yosys #2583: Crashes without an error message

fraser125, 2021-02-11:
## Steps to reproduce the issue
Create a single verilog file called 'crash_test.v' with only the following line
>'include "crash_yosys.h"

alternative [crash_test.zip](https://github.com/YosysHQ/yosys/files/5963417/crash_test.zip)
The actual header file (crash_yosys.h) does exist or cannot be found.

> (Yosys 0.9 (git sha1 1979e0b1, i686-w64-mingw32.static-g++ 5.5.0 -Os) latest download from [yosys download](http://www.clifford.at/yosys/download.html)

1. On Windows 10 Double-Click to open Yosys.exe 
2. type "read_verilog crash_test.v"

## Expected behavior

A parsing error message on the console.

## Actual behavior

Closes the console, immediately.

zachjs, 2021-02-11:
Yosys is not actually crashing here, but exiting gracefully with a parse error. Because `Yosys.exe` has been (implicitly) exec'd, the terminal emulator closes the window upon completion of the process. You can see similar behavior in a standard *nix shell by running `exec yosys crash_test.v`. Some terminal emulators choose instead to leave the window open, but indicate the process has exited.

On the latest master, the error message is
```
Parsing Verilog input from `crash_test.v' to AST representation.
crash_test.v:1: ERROR: syntax error, unexpected OP_CAST
```

While initially poking around for the issue, I noticed at least two problems with Yosys's handling of (potentially recursive) includes. However, it's not clear to me whether this issue intends to refer to problems with includes, or if it instead refers to the surprising terminal behavior described above.

fraser125, 2021-02-11:
My report was about the terminal behavior. 
I've just started learning Verilog and I was trying out a project I found on the web. 
The first couple of files worked fine and remained in the yosys console mode through several 'opt' cycles, then I had 2 in a row that closed the console immediately as described. I narrowed down the cause and since it was so easy to reproduce I thought it would of been an easy issue to overlook for those that already have experience with the yosys workflow.

clairexen, 2021-03-01:
> My report was about the terminal behavior.

This is a choice done by the terminal emulator that Yosys has no control over.

closed 2021-03-01 by clairexen