YosysHQ/yosys #2589: CXXRTL: (* top *) attribute causes wildcard port connections (.*) to break

msinger, 2021-02-16:
## Steps to reproduce the issue

test.sv:
```
(* top *)
module top(
        input  logic       clk,
        output logic [3:0] seq,
    );

    test t(.*);
endmodule

module test(
        input  logic       clk,
        output logic [3:0] seq,
    );

    always_ff @(posedge clk) seq++;
endmodule
```

Generate test.cc with:
```
yosys -p "write_cxxrtl test.cc" test.sv
```

## Expected behavior

The output file test.cc should contain an eval function similar to this:
```
bool p_top::eval() {
    bool converged = true;
    bool posedge_p_clk = this->posedge_p_clk();
    // cells $procdff$3 $flatten\t.$add$test.sv:0$2
    if (posedge_p_clk) {
        p_t_2e_seq.next = add_uu<32>(p_t_2e_seq.curr, value<32>{0x1u}).slice<3,0>().val();
    }
    // connection
    p_seq = p_t_2e_seq.curr;    // <----- IMPORTANT LINE
    return converged;
}
```

## Actual behavior

The output file test.cc does not contain the line marked with the comment "IMPORTANT LINE" in the code snipped above. If the top-attribute is removed from the top module or if the port connections are done more explicit (.clk, .seq) instead of the wildcard, then the output file is correct and works as expected.

I'm not sure, maybe I'm not supposed to assign the top-attribute to the top module manually. In that case I apologize, please just close this issue, I will get the message.

msinger, 2021-02-16:
Sorry, I forgot: I'm not sure if it is a CXXRTL problem specifically. I just wasn't able to reproduce it when using the synth_ice40 command.

zachjs, 2021-02-17:
Wildcard port connections are elaborated as part of the `hierarchy` pass. `write_cxxrtl` automatically invokes `hierarchy -auto-top` if `-nohierarchy` isn't specified and no module has the `top` attribute. Manually running `hierarchy` before `wire_cxxrtl` produces the correct output. It may be possible to change the default behavior of `write_cxxrtl`, but I'm not sure what is intended there.

whitequark, 2021-02-18:
`write_cxxrtl` is doing too much. The part that runs passes should be split off under a name like `synth_cxxrtl`. Until then, @zachjs's workaround is what I recommend.

msinger, 2021-02-18:
Thank you guys. I didn't know that `hierarchy` makes the port connections and that `write_cxxrtl` doesn't run `hierarchy` when the attribute is used. This clears things up.

PR #2874 (2021-07-16): cxxrtl: run hierarchy pass regardless of (*top*) attribute presence
The hierarchy pass does a lot more than just finding the top module, mainly resolving implicit (positional, wildcard) module connections.

Fixes #2589.
commit 5b003d6e (2021-07-16): cxxrtl: run hierarchy pass regardless of (*top*) attribute presence.

The hierarchy pass does a lot more than just finding the top module,
mainly resolving implicit (positional, wildcard) module connections.

Fixes #2589.
closed 2021-07-16 by whitequark